joan2937/pigpio #163: libpigpio memory

MaitreDede, 2017-09-27:
Hi,
I have coded a C# app that uses `libpigpio`. I have crashed due to unhandled signals, and I am searching for the cause. It can be either a C# lib I am using (ImageSharp) or libpigpio.
How can I diagnose if it comes from libpigpio ?

I have errors like that : 
```
2017-09-27 00:00:42 sigHandler: Unhandled signal 11, terminating
2017-09-27 00:00:42 sigHandler: Unhandled signal 11, terminating
*** Error in `./App': double free or corruption (out): 0x6eb00470 ***
2017-09-27 00:00:42 sigHandler: Unhandled signal 6, terminating
```
I have also `Unhandled signal 34` occuring.
These errors appear randomly...

guymcswain, 2017-09-27:
According to similar reported issues like [this](https://www.raspberrypi.org/forums/viewtopic.php?f=81&t=77337)  there may be an uninitialized pointer in your code somewhere.  Gdb may be your friend but I'm no expert.

MaitreDede, 2017-09-27:
Thanks @guymcswain, but I remember one important thing that could be interesting. I am using `libpigpio` cross-thread... I initialize the library in the main thread, and its methods are called from a secondary thread. I don't have concurrency problems in my code (yet), only the lib that is initialized in a thread and use in another. Only 2 threads.
Library is well initialized (gpioInitialise), and work well for some seconds before crashes occure.

The topic you linked looks similar. It not java, but C# in my case... I will try to increase libpigpio logging level to see if I can get more data...

MaitreDede, 2017-09-27:
Well I think there are different things.
First, it seams that the pigpio signal handler can interfere with signals that can be used by the application. For example, if I remember well, Mono uses SIGSEGV signals to handle NullReferenceException. ([docs](http://www.mono-project.com/docs/debug+profile/debug/#debugging-with-gdb)).
So I suspect that the dotnet core runtime for linux also uses signals, but seams they are caught by libpigpio.
If I read the code correctly, signals are only used in logging level changing ([code](https://github.com/joan2937/pigpio/blob/1aa4cca8a69d4f8f8ce198178dbd7f436b4f503c/pigpio.c#L5465-L5521)). Beside the logging, is pigpio signal handling really necessary when used as library ? Logging level could be handled by methods (get/set logging level, and setting a custom logging handler)...

MaitreDede, 2017-09-27:
Just seen https://github.com/joan2937/pigpio/issues/138... That explains things. I will continue in this issue.

closed 2017-09-27 by MaitreDede