joan2937/pigpio #165: Fixes suggested for inclusion in V65

joan2937, 2017-09-28:
I seem to have some time and would like to do a release incorporating fixes for non-contentious issues and pull requests, i.e. clear bug fixes or improvements unlikely to have negative size effects.

I am rather out of touch with issues so if you know any candidates could you reference them below.

guymcswain, 2017-09-28:
Welcome back!

- PR #164 seems to affect many folks (issue #138)  Not all those affected have yet responded.
- Issue #157 affects myself and @Joey-1970.  There could be contention on exactly how it is done.  I would think running `initInitialise()` from a new socket command is sufficient.  From my perspective it's not a critical issue but a strongly desired feature when you are testing a buggy client application.
- Issue #148 will have a PR after completing more testing.  (I'll try to hurry it along)
- Issue #126 is one I really would like to see 'fixed' but I know it's contentious so not for now.
- Several issues are related to folks needing information or clarification on how pigpio works.  I would recommend a wiki be created and somehow allow controlled access to make updates.

PS  Great library!  Let us know how we can help you.

Joey-1970, 2017-09-29:
...one more thing: ;-)
If I use the notification, it is very difficult to split the incomming data. If I look only for the first 32-Bit:
- when it is a command the value must between 0 and (actually) 116
- when I get a Watchdog-Event, a Keep-Alive-Signal or a Event-Signal, I can see this, in the flags, because of the flags-position in this first 32-bit it is much higher then 116
- but in case of a "normal" notification, no flags will be set and the seq-nr can be lower then 116
If I read the second 32-bit it would be clear what it is, because I only use NOIB over this socket and it must be zero after 99 for the first 32-bit, but not the tick, it will be higher..

I wish, that I can see that it is a "normal" noticication"... ;-) 

Joachim

guymcswain, 2017-10-05:
@joan2937 Can you please give us an update on what your plans are for V65?

closed 2020-01-02 by guymcswain