joan2937/pigpio #166: gpioSerialRead() returning corrupt data at baud rate >= 115200

guymcswain, 2017-09-29:
Using socket interface with pigpio v64 and `sudo pigpiod -s1`, this script demonstrates the problem.  I get a failure frequently in under 50 iterations but I've seen it take much longer on some occasions.
```
#!/usr/bin/env/python
import time
import pigpio
from loremipsum import get_paragraph

TXRX = 14

pi = pigpio.pi('10.0.0.236')
if not pi.connected:
  print "can't connect, exiting"
  exit(0)
print 'connected to pi @ 10.0.0.236'

h = pi.serial_open("/dev/ttyAMA0", 115200)

pigpio.exceptions = False
status = pi.bb_serial_read_open(TXRX, 115200)
if status < 0:
  if status == -50:
    pi.bb_serial_read_close(TXRX)
    pi.serial_close(h)
    print 'GPIO in use, try again'
    exit(0)
  else:
    print 'bb_serial_read_open error: {}'.format(status)
    exit(0)
pigpio.exceptions = True

iter = 0
while True:
  iter += 1
  print 'Iteration: ', iter
  p = get_paragraph()
  send = bytearray(p.encode('utf8'))
  print 'send length = {}'.format(len(send))
  pi.serial_write(h, send)
  readBytes = 0
  recv = bytearray()
  while readBytes < len(send):
    time.sleep(0.2)
    (count, bytes) = pi.bb_serial_read(TXRX)
    readBytes += count
    if count: recv += bytes
  print 'read {} bytes'.format(len(recv))
  if recv != send:
    print 'FAIL'
    break
  else:
    print 'PASS'
    
print recv
print send

pi.bb_serial_read_close(TXRX)
pi.serial_close(h)
pi.stop()

print "Goodbye"
```

guymcswain, 2018-05-07:
Hi Joan, were you able to run this script and reproduce the issue?  I have not tried with the latest versions of pigpio unless you think that would be worthwhile.  Let me know if you need additional information.

joan2937, 2018-05-11:
Is this bit bang serial?  I think my tests showed 19.2k or less was pretty solid with anything higher needing checksumming.

guymcswain, 2018-05-11:
It is bit bang.  I'll test with 19.2k.

guymcswain, 2018-08-24:
My apologies for not closing this earlier but I was a) busy, but more importantly b) looking for an explanation for the failures.  I think I have found that explanation in issue #35.  My take away from that issue is, there can be a sampling uncertainty of up to 4us (for 99.9967% of the time).   For 115.2K baud, the bit period is 8.6us and the sample point is half the period or 4.3us.  At that point there is no margin for handling clock error between transmitting and receiving units.

If I've got this wrong I'd appreciate a comment.

closed 2018-08-24 by guymcswain