joan2937/pigpio #167: incorrect read w/10bit addr and clock stretching

HunterWare, 2017-10-04:
I'm trying to do a 10bit addressed read on a device that requires a delay between the read and the response data.  This is definitely a bit of a corner case for pigpio but I think that the underlying support is present (which is really cool).

The read command is Read(0x055) D1.  I implement it in bb_i2c_zip as:
(count, data) = pi.bb_i2c_zip(SDA, [4, 0x78, 2, 7, 1, 0x55, 4, 0x78, 2, 6, 1, 3])

Effectively I setup a 1 byte write of 0x55 Write to 7bit addr 0x78, send a START, then read from 0x78 of 1 byte.  I think this ends up correctly emulating a proper 10bit address read, and the scope looks right, but the returned value given to me is wrong and pigpio is clocking in more than receive 1 byte.

Here is the setup for the read on scope...
[image]

Here is the response being clocked in after the stretched SCL.  Notice more than one byte is clocked, and the returned value isn't what is seen on scope...
[image]

The call above actually returns: count:1 data:[e0]

A couple things that might be relevant:
- I have other 7bit non-stretched things communicating properly using basically this setup.
- I am using GPIO 2,3 (HW I2C disabled) so that the pulls are setup properly in HW.
- Init is pi.bb_i2c_open(2, 3, 1000).

Thank you,
  Hunter

HunterWare, 2017-10-07:
Work around: If you terminate every transaction with a "0", then things operate correctly.  In the example for bb_i2c_zip I see multiple transactions without a "0" after each stop, but in my case I do not get proper results without terminating each transaction with "0".

guymcswain, 2020-01-02:
Add this to the project's Wiki under available work-arounds.

guymcswain, 2020-01-25:
From the pigpio documentation:

> The address, read, and write commands take a parameter P. Normally P is one byte (0-255). If the command is preceded by the Escape command then P is two bytes (0-65535, least significant byte first).

Therefore, the proper way to do 10-bit addressing, using your example, would be:
`(count, data) = pi.bb_i2c_zip(SDA, [1, 4, 0x78, 0x55, 2, 6, 1, 3, 0])`

> The address and flags maintain their previous value until updated.

So, no need to repeat the address.

Finally, the message packet should always be terminated with 0 (End).  This is shown in the documentation examples.

closed 2020-01-25 by guymcswain