joan2937/pigpio #168: Very fast read after interupt of PCA9655E

Joey-1970, 2017-10-04:
Hello,
again I search for a good idea.
I use a PCA9655E as input, no problem to read over socket interface after a interrupt show a change on the inputs. But I must press the button around 250ms to get the right information because 
- PIGPIO see interrupt
- notify my application
- I send a read-command to PIGPIO
- PIGPIO read the status
...but if the button pressed to short, the real value is gone...

I think writing a PIGPIO-script could be very much faster. 
- Start script when interrupt go to 0 (wait x)
- read input value (I2CRW handle register)
...but how I get this input-value thatis "stored" there?
Is there a way to send it via socket interface?
Can I read this result from the socket interface?

Joachim

dbbnrl, 2017-11-29:
I don't entirely understand your application or how you're interacting with the socket API, but I'll give it a shot.

It seems to me that if you use a script as you propose, you then have two problems in your client code:

- Figure out *when* the interrupt occurred
- Figure out *what* value was read by the script when the interrupt occurred.

Let's address the *what* first.  The `script_status` command (PROCP) returns, in addition to a status code, the current values of all 10 "parameter" registers used by the script.  As long as your script stores the result of its read in p0-p9, you can observe these results by calling `script_status`.

As for the *when*, the simplest solution is probably to have your client register for the same interrupt that your script waits on.  By the time the client receives the notification and sends a `script_status` command to the server, the script should have finished performing the read and storing it in a parameter register.

All of this will work reasonably only for relatively infrequent interrupts.  If multiple interrupts might occur in the time it takes for the client to be notified, send a query, and get a response, you'll lose events.

In that case you might have some luck looking into the "event"-related commands.  If you only have a few GPIOs that you're monitoring, you could use `event_trigger` command (EVT) in your script.  On the client end, you would use `event_callback` or `wait_for_event` to be notified about events, rather than GPIOs.  You'd have to assign meaning to a very small set of events (i.e. maybe event 0 means "GPIO 5 falling" and event 1 means "GPIO 5 rising").

This should pretty much guarantee you never lose information, at the cost of supporting only a small number of GPIO pins.

guymcswain, 2017-12-31:
Just be aware that EVT in script has a bug.  The following quote is a reply from joan re issue #148.
> There is an error within the pigpio scripts code which will prevent the use of event trigger (EVT). The code checks for internal/external commands by testing against command number 100 (internal commands start at 800). The constant 100 was never a great choice but didn't seem so bad when the maximum command number was in the 90s.
A temporary fix is to change a line in the pthScript function in pigpio.c (currently line 6548).
Change if (instr.p[0] < 100) to if (instr.p[0] < PI_CMD_SCRIPT).
When I have time I will tidy up scripts a little by preventing commands which have no useful functions within the script.

joan2937, 2018-02-13:
The change from "if (instr.p[0] < 100)" to "if (instr.p[0] < PI_CMD_SCRIPT)" will be in V66.

closed 2018-02-20 by joan2937