joan2937/pigpio #169: Using GPCLK2  results in broken putty and vncviewer

pa3gsb, 2017-10-08:
Hi,

context RPI 3
- gpiod v 60
- uname -a : Linux raspberrypi 4.4.50-v7+ #970 

Using simple program (see below) results in broken putty connection (not possible to recover) and also nvncviewer is loosing the connection.

Some ideas?

Johan

#include <stdlib.h>
#include <stdio.h>
#include <pigpio.h>

int main(int argc, char **argv)
{
	if (gpioInitialise() < 0) {
		fprintf(stderr,"gpio could not be initialized. \n");
		exit(-1);
	}
	
	gpioSetMode(6, PI_ALT0);
	
	int result = gpioHardwareClock(6, 10000000);

		
	gpioTerminate();
}

guymcswain, 2017-10-08:
result >= 0?
Does it work on other RPi (except model A, B)?

pa3gsb, 2017-10-09:
Hi

result =0.

Both wifi and ethernet connections are gone.

Using mouse and keyboard and screen, raspberry can be controlled. 

i do have a pi 2 but did not try it yet

pa3gsb, 2017-10-09:
RPI 2 it is working.

Same linux version and pigpio .

guymcswain, 2020-01-02:
Please retest with the latest version.

pa3gsb, 2020-01-03:
Solved

closed 2020-01-03 by pa3gsb