joan2937/pigpio #170: Bug or Feature? Client Socket that are closed from remote application won't close from PIGPIO

Joey-1970, 2017-10-09:
Hello Joan,

I don't know if you read about it, but it is a problem if you have a application that will always open a new socket, do something and close the socket. In PIGPIO the mem raise from around 11.000 at the start to above 3.000.000 short before PIGPIO won't take further socket connection.

Joachim

guymcswain, 2018-09-11:
@joan2937 ,

Would you consider adding changes to pigpio socket handlers to clear out 'dead peer' connections?  I encounter this issue during early debug phases where my application crashes frequently or otherwise gets shutdown ungracefully.  I have to periodically remember to reboot my RPi running the pigpio daemon.

More importantly, if as suggested by @Joey-1970 , the connections remain open even after the peer socket is closed gracefully, this issue looms for applications that want to run reliably over the long term but may reboot periodically to 'clean house.'

guymcswain, 2018-09-12:
Looking at the source code reveals that sockets are closed when socket receives return value -1 (error) or return value 0 (peer performed an orderly shutdown).  So this issue applies only to 'dead peers'.

```
static void *pthSocketThreadHandler(void *fdC)
{
   // ...
   while (1)
   {
      if (recv(sock, p, 16, MSG_WAITALL) != 16) break;
      // …
   }
   closeOrphanedNotifications(-1, sock);
   close(sock);
   return 0;
}
```

guymcswain, 2020-01-02:
Marking this issue as invalid as originally written.  Pigpio closes connections and frees resources when requested.  You may be seeing a corner case where your application is opening then closing sockets in a tight loop before the operating system has time to release the resources.

Dead peer connections - not stated in original issue - are handled since V68.

closed 2020-01-02 by guymcswain