joan2937/pigpio #171: Starting pigpiod without sudo?

Nic2012, 2017-10-15:
Hi,
trying a special router distro for the RaspberryPi 3 from LEDE the latest 17.01.3,
https://lede-project.org/docs/instructionset/arm_cortex-a53_neon-vfpv4

Compiled like
` CFLAGS=-mfloat-abi=hard make` and installed.

Trying to run gives me:
```
root@lede:~/pigpio# /lib/ld-musl-armhf.so.1 ./pigpiod
root@lede:~/pigpio# 2017-10-15 08:00:27 initCheckPermitted:
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+

Can't initialise pigpio library
```
Traditionally a router os has no sudo on board, but there was an add. package "sudo" that I have installed but it makes no difference. Is there any chance to startup the daemon?

Any idea would be appreciated.
Best regards,
Nic

MaitreDede, 2017-10-20:
Hi @Nic2012 ,
Pigpio uses direct memory access that needs the root privileges. 
First, you can check if direct memory access is available : is `/dev/mem` present ?
Then the user that runs the application needs to have access to `/dev/mem`.

joan2937, 2017-12-20:
Answered by @MaitreDede

closed 2017-12-20 by joan2937