joan2937/pigpio #172: NOOB - pigpio seems to work but doesn't actually change GPIO pins

MisterClarkRGC, 2017-10-16:
This sure seems like a noob question but I've a couple of days searching for a solution and I've come up empty. I believe I am following every instruction I can find correctly and have run out of things to google.

**My task**
I want to use a pi to independently control multiple motors with PWM.  I believe pigpio is a good way to do that as the pi only has a single PWM hardware output.  To test / learn this code I am currently attempting to make some LEDs on a breadboard pulse.  The thing is when I copy very simple code that others say works for them, everything runs fine but the LEDs never change.  I suspect pigpio is not connected to the hardware (no idea how to test this though).

**What I've done so far**
I have pigpio installed as I have updated to Raspian-(jessie) which comes with it by default.  
I've chosen to use pin 16 aka GPIO no 23, the pin, it is connected to an LED with a 330 Ohm resistor. The LED is the correct way round and the other leg is connected to ground (pin 6). I think there is nothing wrong with the pin as I can turn on the LED with a different code using the standard RPi.GPIO commands without pigpio.  

I believe I have (eventually) gotten the pigpiod deamon to run properly via the crontab at boot.  I believe it is running fine because I don't get the long warning stating that it isn't running anymore.  What I have done for this is add the line (without quotes) "@reboot sudo      /usr/bin/pigpiod" to the crontab. Note that this is slightly different than the basic instruction because pigpiod is located in that folder for RPi 3 Model B apparently. I was initially using /usr/bin/pigpiod which wasn't found when running code. 

**My code** (sorry don't know how to format with colours etc)- 

   import time
   import pigpio # imported the library 

   pi = pigio.pi()    # given access to the GPIOs 
   pi.set_mode(16,pigpio.OUTPUT)      # I've then set pin 16 to be an output using   
   print(pi.connected) # then confirmed a connection with (returned as 1) 
   pi.write(16,1) #  tried to turn on the LED by turning that pin high with 
   time.sleep(2) # delay added to allow me to see LED turn on

The LED does not turn on
   print(pi.read(16)) # I've checked that the write command is doing something with this, it prints 1
   pi.stop() # stop pigpio

PS - other info
What I intending to use was PWM with
set_PWM_frequency(16,10) # choose low freq for testing (so I can see flicker)
pi.set_PWM_dutycycle(16,100) # mess with 2nd argument to change % of time it is on for

 the VEX motor controllers I want to use states that a pulse length of 1ms is full forwards and 2ms is full reverse. 1.5ms is neutral. That is all the data I have.  

I'm just guessing from the code I've read that I should not be using PRi.GPIO in combination with pigpio.  I've tried it with and without in a range of combinations and either way the pigpio commands don't do anything to the pins.

I have also tried running my code with sudo from the terminal
I can copy the actual code over from my pi or post a photo of the circuit if need be.

guymcswain, 2017-10-17:
Pigpio uses the GPIO numbers so try pi.write(23, 1) and of course pi.set_mode(23,...).   I have not used PWM api.  Good luck.

MisterClarkRGC, 2017-10-17:
Thanks, that fixed my problem. Closing thread

Re servo
I found pigpio looking for PWM, I read about Servo when trying to get pigpio running, I suspect that Servo is probably what I need but to be honest I'm not yet sure what the difference is.  We'll read about the differences and likely try both once we have the motor and controller connected to a battery and the pi.

closed 2017-10-17 by MisterClarkRGC