joan2937/pigpio #174: PIGPIO-Crash after WPA-Update?

Joey-1970, 2017-10-25:
Hello,

I see on two Raspberry Pi that are connect over WLAN that sometimes the PIOGPIO-service stops. I can start it again, but after a few hours it is down again. I can't observe it on Raspberry Pis that are connect over LAN.
I few days ago, there was a WPA update, maybe that will caused this?

Joachim

guymcswain, 2017-10-26:
Hi Joachim,
What actions/observations indicate to you that "the PIGPIO-service stops?"
Guy

Joey-1970, 2017-10-26:
Hello Guy,

I saw in my application, that the port 8888 is not available. So I look at the Raspberr Pi with "top" and I don't see "pigpiod". I start it again and it run (and I see pigpiod in "top"), but only for a few hours/minutes, then pigpiod is gone again. A friend of mine call and tell me the same story - again on a Raspberry Pi that is connected over WLAN, I don't observe it on LAN...

Joachim

joan2937, 2018-02-13:
If you can provide a test script I will try this out.  Otherwise I will assume it is a temporary abberation.

closed 2018-02-13 by joan2937