joan2937/pigpio #175: pigpiod - can not start daemon, mmap gpio failed

michalseidl, 2017-11-03:
Hello I have RPI B+ with raspibian installed from 
 ```
https://github.com/debian-pi/raspbian-ua-netinst/releases/tag/v1.0.9
raspbian Linux pi 4.9.0-4-rpi #1 Raspbian 4.9.51-1+rpi1~jessie (2017-10-12) armv6l GNU/Linux 
apt sourece list deb http://mirrordirector.raspbian.org/raspbian jessie main firmware
deb http://archive.raspberrypi.org/debian jessie main
```
and I can not start pigpiod with result both as root or with sudo

```
2017-11-03 13:48:40 initPeripherals: mmap gpio failed (Operation not permitted)
Can't initialise pigpio library
```
The same happens for pigpiod from distribution and compiled version.

Any suggestions? Thanks Michal

michalseidl, 2017-11-10:
Hello,
I have reinstalled RPI with NOOBS lite v 2.4 with minimal Raspbian and everything looks OK and problem with pigpiod has gone.
```
Linux visco 4.9.41+ #1023 Tue Aug 8 15:47:12 BST 2017 armv6l GNU/Linux
deb http://mirrordirector.raspbian.org/raspbian/ stretch main contrib non-free rpi
```

Michal

eras, 2017-11-14:
I have the same problem and I imagine ti is due to my kernel version: 4.9.51-1+rpi3.

jsilence, 2017-12-29:
I have the same error on TinkerOS 2.0.4 on the ASUS Tinkerboard.

`Description:    Debian GNU/Linux 9.3 (stretch)`
`Linux tinkerboard 4.4.71+ #1 SMP Wed Nov 15 14:31:48 CST 2017 armv7l GNU/Linux`

fivdi, 2017-12-29:
@jsilence pigpio will only function on a Raspberry Pi. It will not work on an ASUS Tinker Board.

closed 2018-02-08 by joan2937

bobpitt01, 2018-11-05:
I have the same problem with my Pi Model 3B+ running opensuse Tumbleweed.  pigpiod works fine though on the same pi when running RASPBIAN.  What files in /dev does pigpiod use?  Under opensuse the only *gpio* file is /dev/gpiochip0, where as under RASPBIAN there are three:
crw-rw---- 1 root gpio 254, 0 Aug 24 11:17 gpiochip0
crw-rw---- 1 root gpio 254, 1 Aug 24 11:17 gpiochip1
crw-rw---- 1 root gpio 248, 0 Aug 24 11:17 gpiomem
Thanks for your assistance.