joan2937/pigpio #176: GPIO serial write

gorogm, 2017-11-05:
Dear Joan / devs,
I respect the work you put in this great library!
Let me ask about gpio (bitbang) serial writing functionality: is it supported by the library? I couldn't find it in the documentation / header, although writing functionality appears in bitbang i2c, spi.
Thanks

joan2937, 2017-11-05:
For C see http://abyz.me.uk/rpi/pigpio/cif.html#gpioSerialReadOpen etc,  There should be similar calls for Python, pigs, etc.

joan2937, 2017-11-05:
I closed this by mistake.

gorogm, 2017-11-05:
Thanks Joan for the quick answer, but I would like to WRITE, however all these gpioSerial* function show only READING functionality. Even the opener is 'gpioSerialReadOpen', and I couldn't find a writer counterpart. Do I miss something? Thanks!

joan2937, 2017-11-05:
gpioWaveAddSerial

gorogm, 2017-11-06:
Ahhh great, thank you!

closed 2017-11-06 by gorogm

joan2937, 2017-11-06:
My mistake.  I had forgotten that the write was different to the read (for reasons which are apparent when you write the code, I perhaps should link between the calls in the documentation).

gorogm, 2017-11-06:
That would be helpful for future users! Anyhow, thanks for the quick help!!!