joan2937/pigpio #177: Using gpio>31 for PWM

j123b567, 2017-11-10:
We are using RPi CM3 and so, there is possible to use almost all 53 GPIOs.
I would like to use PWM on gpio>31 but it is not supported by gpiod.

There is `PI_MAX_USER_GPIO` define limiting the number to 32. There are also many `XyzBits` variables using datatype `uint32_t`.

If we for example rise the limit, change the data type to `uint64_t` and fix all constructions like `1<<gpio`. Can it work, or are there other limitations/obstacles to solve?
E.g. it will have performace impact, change the protocol format, ...?

joan2937, 2018-02-08:
It's not a trivial change because the DMA  chain must be extended to allow writes to both GPIO registers.  This was implemented in the very first versions of pigpio but has long been removed as pointless on all but the compute module. 

It's not something I will add back in.

If you would like to see the early code let me know and I'll see what I can find.

joan2937, 2018-02-09:
I'm not sure I have ever suggested that pigpio supports the compute module completely.  The most I remember saying is that pigpio works on all Pi models.

Look at http://abyz.me.uk/rpi/pigpio/index.html and notice the repeated mention of GPIO 0-31.

SlySven, 2018-02-12:
Well, that link says, amongst other things:
# The pigpio library
## Features
...
* reading/writing all of the GPIO in a bank (0-31, 32-53) as a single operation

...

### GPIO
ALL GPIO are identified by their Broadcom number.  See elinux.org
There are 54 GPIO in total, arranged in two banks.

Bank 1 contains GPIO 0-31.  Bank 2 contains GPIO 32-53.
There are several types of board, each with different expansion headers, giving physical access to different GPIO.  
* Type 1 - Model B (original model)
* Type 2 - Model A, B (revision 2)
* Type 3 - Model A+, B+, Pi Zero, Pi2B, Pi3B
* Compute Module - Compute Module

...
#### Compute Module

All 54 GPIO may be physically accessed.  Some are reserved for system use - refer to the Compute Module documentation.
For all types of Pi it is safe to read the GPIO. If you try to write a system GPIO or change its mode you can crash the Pi or corrupt the data on the SD card.

_I suggest it could be emphasised that **pigpio** only supports the **first** GPIO bank which is sufficent for all RPi models **except** the CM series (a mention for the Compute Module 3 and 3 Lite ought to go into that list though I do not know what hardware numbers *they* have)_

_OTOH I appreciate the effort you have put into this project and this is only about part of the library so I am not going to worry about this as it doesn't affect me currently <sup><small>1</small></sup> - it is just that handling the corner cases are what marks out an excellent product._ 😀

\<footnote> 1 - I'm still keeping my eyes open for a cheap CM/CM3 dev kit on a well known on-line auction site as that is the expensive part for the home hobbyist to get their hands on - so if I ever have success I'll be sure to give an update on the hardware numbers...\</footnote>

guymcswain, 2021-01-02:
I'm re-opening this issue to remind me to update the documentation. 

APIs affecting GPIO 32-53:
- gpioRead()
- gpioWrite()
- gpioRead_Bits_32-53()
- gpioWrite_Bits_32_53_Clear()
- gpioWrite_Bits_32_53_Set()
- setMode()
- getMode()
- setPullupDown()
- setISRfunc()