joan2937/pigpio #178: Proposed: Allow parameter changes to running scripts

dbbnrl, 2017-11-27:
I've been playing around with implementing closed-loop speed control of motors in pigpio script (tons of fun; brings back my assembly language days).  It's very useful to use PROCP on a running script to monitor its status (how far has the motor moved?).  I think it would open up some interesting possibilities if the reverse communication direction were possible:  Allow the user to overwrite (some of the) parameters of a script while the script is running.  In my application, for instance, this would allow run-time adjustment of the desired speed without the glitch of stopping & restarting the script.

I can think of two possible implementation approaches:

1. Add a new command (PROCU?) called similarly to PROCR.  Unlike PROCR, the script would not need to be in the PI_SCRIPT_INITING state.  PROCU would alter the value of the first n parameters (however many are passed to PROCU).
2. Instead of adding a new command, modify PROCR so it can be called on a running script.

The implementation would be complicated a little bit by dealing with race conditions, and I admit I haven't thought through all the details yet.  It might be necessary to double-buffer the parameter updates and add a new script-only command to act as a synchronization point (this command would lock the mutex, copy any "pending" parameter updates over the real parameters, and unlock the mutex).

I'd be interested in developing this feature and submitting a PR, if it seems like something that would be accepted/useful.  Thoughts?

Joey-1970, 2017-11-29:
Hello dbbnrl,

if you have so much fun, maybe ou can tell me how I get back data/results from the script via socket (iisue #168)? ;-)

Joachim

dbbnrl, 2017-11-29:
I'll reply in #168.

joan2937, 2018-02-12:
This will be added to V66.

PI_CMD_PROCU
gpioUpdateScript (C)
update_script (Python, pdif2)
PROCU (pigs)

closed 2018-02-20 by joan2937