joan2937/pigpio #180: Wrong SPI speed using Raspberry Pi 3

jmmac1973, 2017-12-07:
Hi,
I'm using the PIGPIO library for SPI communications and everything worked well in Raspberry Pi 1 (RPi as master at 10MHz). But then I switched to Raspberry Pi 3 and the SPI speed changed from 10MHz to 16MHz.

I believe the issue comes from the different VideoCore bus speeds, 250MHz in RPi1 and RPi2 and 400MHz in RPi3 (16/10 = 400/250). Maybe PIGPIO doesn't take that into consideration when calculating the clock divider ?

Jaime

joan2937, 2017-12-08:
I believe this is the same root issue as that first mentioned in #80 and #140.

I haven't been looking very closely at posts and do not remember seeing any solution.

jmmac1973, 2017-12-11:
The following page is relevant to the issue and offers a solution (that worked for me).

https://github.com/raspberrypi/linux/issues/2094

Solution: add 'core_freq=250' into /boot/config.txt/

guymcswain, 2020-01-02:
If your setup is headless (no monitor) and you have no audio configurations please report back that status.  Otherwise I'm tagging this as an hdmi/audio interaction issue with available work-around.

jmmac1973, 2020-01-03:
I'm no longer working on that project but my setup was headless. I didn't used audio but can't confirm my audio configurations. Probably I removed everything not essential to the project.

closed 2020-03-22 by guymcswain