joan2937/pigpio #183: Installing pigpio on virtual environment

craver2000, 2017-12-10:
Hi, I wanted to ask if it is possible to use pigpio in a virtual environment? 

Running the servo demo script worked for me in a non-virtual environment with the command 'sudo python servo_demo.py' When I activate a virtual env and proceed with the same installation and activation steps, I am likewise able to use the demo script with the same command.

However, the problem arise if I leave out 'sudo'in my bash commands (i.e. python myscript.py' ) since using the command 'sudo python myscript.py' would render another module unavailable in the virtual environment.

The error message when I used the non-sudo command goes:
Traceback (most recent call last):
  File "servo_demo.py", line 13, in <module>
    import pigpio
ImportError: No module named pigpio

I appreciate any advice on how I can run pigpio module without using 'sudo' in a virtual environment.

guymcswain, 2020-01-02:
Need a Python expert to assist with this question.

dir-ableton, 2020-03-08:
While it would be nicer if the `pigpio.py` was packaged on PyPI for pip-assisted installation, it's actually very easy to install `pigpio.py` yourself. Just copy it into your venv's site-packages. You can find these under `{sys.prefix}/lib/pythonX.X/site-packages`, and to get at the prefix just run

`python -c 'import sys; print(sys.prefix)'`

joan2937, 2020-03-08:
Ben Nuttall maintains a pigpio version on PyPy.

https://pypi.org/project/pigpio/

I do not understand how this is relevant to virtual environments.

Neither do I understand what you want us to do to "solve" this issue.

dir-ableton, 2020-03-09:
It can be depending on the way the venv has been set up and the original poster has set it up and installed pigpio. There is for example the option to isolate venvs from system-installed packages, and that could have come into play here.