joan2937/pigpio #184: Secure socket interface

doragasu, 2017-12-19:
First of all, thanks for the library, it's really useful and easy to use. I have developed an application that controls GPIO pins from a separate PC in no time.

Currently my only issue with this library is about security. As far as I know, the socket interface does not support even a simple authentication stage.

Are there any plans for securing this interface?

joan2937, 2017-12-20:
I have no plans to add any security.  It is not my domain and anything I did would be false security at best.

If anyone knows how to properly achieve a level of security it would be nice to know.

MaitreDede, 2017-12-20:
As a workaround, it could be easy to create a side app I may call "proxy", that expose the "secured" socket, and redirect data flow to local (or remote) pigpio. For example (that does not do ssl) : https://www.npmjs.com/package/node-tcp-proxy

doragasu, 2017-12-21:
The proxy could work, but maybe it adds too much complexity.

I have neither a security background, but I presume that the best way could be using TLS with public/private certificate pairs.

In the meantime, maybe a simple authentication function and IP based restrictions, although not secure, could provide a minimum amount of security. Something like this using a pre-shared key could be enough:
1.- Upon connection, commands are rejected unless client authenticates.
2.- Client requests authentication. As a response, pigpio sends a big random number.
3.- Client receives the number, adds the pre-shared key and performs e.g. a SHA256 of the combination. Then sends the result to server.
4.- Server performs the same operation and compares the results. If OK, allows further commands to succeed.

I'm sure security experts will find a lot of flaws in this proposal, but for me something like this  could be enough.

damienstuart, 2017-12-30:
You could also look at using stunnel (https://www.stunnel.org/index.html).  It is a proxy designed to add TLS encryption functionality to existing clients and servers.

joan2937, 2018-02-08:
I am not planning any work in this area.

Please reopen the issue if you know of a workable solution.

closed 2018-02-08 by joan2937