joan2937/pigpio #185: No inputs mode

BWAvants, 2017-12-31:
Hey Joan,

First, thanks so much for this library!  Although I may have missed something in my research before picking up pigpio, it is the only library available that could enable my project.

I would very much like to be able to turn off the pin polling and logging (essentially all the inputs) either permanently or selectively during operation.

At present, once pigpiod is loaded it is continually taking resources... even at the 10us sample rate on my Pi3 it averages more than 6.5% usage on a core.  I have seen a similar request (sometime in 2015) that you ultimately turned down, but I believe there are use cases for your library that don't require inputs but still cannot be done with another option.  Also there are instances where inputs are only necessary periodically, but killing and restarting pigpiod with root privileges isn't a good option.

In my personal case, the functionality I need is the wave transmit system.  I have made a stepper controller (along with a flask front-end server and a gphoto2 DSLR camera interface) that implements smoothed acceleration and deceleration, but is responsive to command changes in a way that seems instant.  I have not found another python method to accomplish this at higher stepping rates (microstepping at a decent clip).  With pigpiod, I accomplish it by pre-computing waves of ~25ms worth of pulses, which allows a fairly rapid response to a command or input while still allowing both precise timing and step counts.  At any rate, I will likely need to run this setup off of batteries at some point and I'm trying to minimize power usage by leaving the system idle as much as possible when not directly in use.  If you still don't think this is a good feature for pigpio, do you know of another library with similar DMA control (ie able to stream an acceleration curve in small chunks)?  I should mention that I'm developing the whole thing in python3.

Thanks again for pigpio!

guymcswain, 2017-12-31:
Does this work for you?  #132

BWAvants, 2017-12-31:
Yes, this would be what I'm looking for!  Thanks for the pointer, I don't
think I would have found it myself.  =]

- Ben Avants
* Robinson Lab*
* Rice University*
 (713) 348-2978

joan2937, 2018-02-08:
Should be fixed in V65.

closed 2018-02-08 by joan2937