joan2937/pigpio #187: gpioInitialise hangs on 64 bit Debian on Pi 3 (pi64)

ZiCog, 2018-01-05:
I am using pi64 on a Pi 3. https://github.com/bamarni/pi64

After "make" and "make install" I find gpioInitialise() never returns.

This is the simple code that hangs:
```
#include <stdio.h>
#include <pigpio.h>

int main (int argc, char* argv[])
{
    if (gpioInitialise() < 0)
    {
        printf("OK");
    }
    else
    {
        printf("FAIL");
    }
    return(0);
}
```
There are a great many compilation warnings produced by make. Mostly regarding the casting of pointers to unsigned.  Like so:
```
gcc -O3 -Wall -pthread -fpic -c -o pigpio.o pigpio.c
pigpio.c: In function 'waveCbPOadr':
pigpio.c:2854:11: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
    return (uint32_t) &dmaOBus[page]->cb[slot];
           ^
pigpio.c: In function 'waveOOLPOadr':
pigpio.c:2885:11: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
    return (uint32_t) &dmaOBus[page]->OOL[slot];
           ^
pigpio.c: In function 'wave2Cbs':
pigpio.c:2999:16: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
    p->src    = (uint32_t) (&dmaOBus[0]->periphData);
                ^
pigpio.c:3078:22: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
             p->src = (uint32_t) (&dmaOBus[0]->periphData);
                      ^
pigpio.c: In function 'i2cReadBlockData':
pigpio.c:3639:60: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
    DBG(DBG_USER, "handle=%d reg=%d buf=%08X", handle, reg, (unsigned)buf);
                                                            ^
pigpio.c:208:46: note: in definition of macro 'DO_DBG'
             myTimeStamp(), __FUNCTION__ , ## arg);                 \
                                              ^~~
pigpio.c:3639:4: note: in expansion of macro 'DBG'
    DBG(DBG_USER, "handle=%d reg=%d buf=%08X", handle, reg, (unsigned)buf);
    ^~~
pigpio.c: In function 'i2cReadI2CBlockData':
pigpio.c:3786:27: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
       handle, reg, count, (unsigned)buf);
                           ^
pigpio.c:208:46: note: in definition of macro 'DO_DBG'
             myTimeStamp(), __FUNCTION__ , ## arg);                 \
                                              ^~~
pigpio.c:3785:4: note: in expansion of macro 'DBG'
    DBG(DBG_USER, "handle=%d reg=%d count=%d buf=%08X",
    ^~~
pigpio.c: In function 'i2cReadDevice':
pigpio.c:3913:22: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
       handle, count, (unsigned)buf);
                      ^
pigpio.c:208:46: note: in definition of macro 'DO_DBG'
             myTimeStamp(), __FUNCTION__ , ## arg);                 \
                                              ^~~
pigpio.c:3912:4: note: in expansion of macro 'DBG'
    DBG(DBG_USER, "handle=%d count=%d buf=%08X",
    ^~~
pigpio.c: In function 'i2cZip':
pigpio.c:4087:48: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
       handle, myBuf2Str(inLen, (char *)inBuf), (int)outBuf, outLen);
...
... 
...
```

joan2937, 2018-01-05:
I can't say I am that surprised.  I have never tried a 64-bit Raspberry system and don't have any plans to do so unless and until standard Raspbian becomes 64-bit.

I'll assume pigpio is incompatible with a 64-bit system.

ZiCog, 2018-01-05:
All those warnings about casting 64 bit pointers into 32 bit integers make me think this has not been written with cross-platform support in mind. Which is fair enough but it leaves me stuck. I have other software to run here as well that is better suited to a 64 bit machine.

joan2937, 2018-01-05:
There is a linguistic clue in pigpio, it is short for (Raspberry) Pi GPIO.  It has never been intended to be cross-platform and the most interesting functionality will only ever work on the Broadcom based Raspberry Pis.

Do not use pigpio if your project is intended to run on multiple platforms.

closed 2018-01-05 by joan2937

ZiCog, 2018-01-05:
Ah yes. I'm totally with you there. I do not expect pigpio to be usable on other hardware.

But I am indeed using real, genuine, Raspberry Pi 3's.

I was only hoping pigpio would work with other Linux distros than just Raspbian. I see no reason why it should not.