joan2937/pigpio #189: Command thread locks can deadlock in certain situations

maihde, 2018-01-30:
The code in `_pigpio_command` and `_pigpio_command_ext` uses `acquire()` and `release()` without a corresponding try/finally.  The lines between these can throw exceptions, such as conditions where the socket connection has been dropped and sl.s.recv returns less information than expected and unpack throws an exception.

The code should look like:

```
sl.l.acquire()
try:
    ...
finally:
    if rl: sl.l.release()
```

maihde, 2018-01-31:
Steps to reproduce:
1. Start `pigpiod` on the RaspPi
2. From a client execute
    
    python
    >>> pi = pigpio.pi('hostname')
    >>> pi.get_current_tick()
    >>> pi.get_current_tick()

3. Kill the `pigpiod` daemon
4. Continue the client execution

    >>> pi.get_current_tick() # <- this throws an exception on struct.unpack
    >>> pi.get_current_tick() # this deadlocks

Further investigation also reveals that there are many places where the `rl` argument is set to false.  This can be dangerous because there are many places where an exception can occur causing a deadlock.  Instead of using the `rl` argument a safer practice is to use `threading.RLock` so that both the outer function (i.e. `_i2c_read_block_data`) and the inner function `_pigpio_command` can use proper acquire/try/finally/release blocks.

A Pull Request is forthcoming to address these issues.

joan2937, 2018-02-02:
The #186 fix fails in Python 3 (may just be a typo, I haven't studied it in detail).

Is your fix a replacement or a complement to #186?  Does it work for Python 2 and Python 3?

maihde, 2018-02-02:
This is a full replacement for #186 with additional protections and improved behavior.  It should run on both Python 2 and 3; I've been using it on version 3 for the past couple days with this code (https://github.com/maihde/hass-rpi-gpiozero) and have encountered no issues.  I'll confirm version 2 compatibility later tonight.

For what it's worth, #186 is failing in Python 3 because it's using

```
try:
    ...
except Exception, e
    ...
```

Instead of:

```
try:
    ...
except Exception as e
   ...
```

joan2937, 2018-02-02:
Yes, I made the change to "Exception as e" to run the code under Python 3.  I wondered if there was some deeper problem.

I have been trying to merge the various pull requests with my current bug fixes.  I will probably revert #186 and try your #189 instead.

maihde, 2018-02-03:
I just had the opportunity and I tested the following:

```
import pigpio
pi = pigpio.pi('hostname')
pi.write(4, 0)
```

With both Python 3.5.2 and Python 2.7.12 and they worked with #189.

joan2937, 2018-02-08:
Should be fixed in V65.

closed 2018-02-08 by joan2937