joan2937/pigpio #191: pigpio error

busrakb, 2018-02-02:
Hi,

I am taking this error sometimes. How can i fix it? Can you help me, please?

%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio at 192.168.1.128(8888)

Can't create callback thread.
Perhaps too many simultaneous pigpio connections.
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/dist-packages/watchdog/observers/api.py", line 199, in run
    self.dispatch_events(self.event_queue, self.timeout)
  File "/usr/local/lib/python2.7/dist-packages/watchdog/observers/api.py", line 368, in dispatch_events
    handler.dispatch(event)
  File "/usr/local/lib/python2.7/dist-packages/watchdog/events.py", line 330, in dispatch
    _method_map[event_type](event)
  File "/home/demo/classifier.py", line 272, in predict
    pi1.write(21,1)
  File "/usr/local/lib/python2.7/dist-packages/pigpio.py", line 1381, in write
    return _u2i(_pigpio_command(self.sl, _PI_CMD_WRITE, gpio, level))
  File "/usr/local/lib/python2.7/dist-packages/pigpio.py", line 983, in _pigpio_command
    sl.s.send(struct.pack('IIII', cmd, p1, p2, 0))
AttributeError: 'NoneType' object has no attribute 'send'

joan2937, 2018-02-02:
I would take the message at face value.

**Perhaps too many simultaneous pigpio connections.**

Perhaps you have lots of open connections.

joan2937, 2018-02-08:
Please reopen this issue if you can provide a small script which demonstrates the problem.

closed 2018-02-08 by joan2937