joan2937/pigpio #195: Pigpio cant connect to localhost(8888)

EXTRAMINER, 2018-02-19:
Hi, i have a raspberry pi3 with hassbian,and this is the second time that this error came, the first time i fixed with a homeassistant update, but now i cant fix with the new update. I think that there is a fix, but i don't know how to fix, can anyone help me, thank you very much.
```
Feb 19 22:01:18 hassbian hass[583]: %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Feb 19 22:01:18 hassbian hass[583]: Can't connect to pigpio at localhost(8888)
Feb 19 22:01:18 hassbian hass[583]: Did you start the pigpio daemon? E.g. sudo pigpiod
Feb 19 22:01:18 hassbian hass[583]: Did you specify the correct Pi host/port in the environment
Feb 19 22:01:18 hassbian hass[583]: variables PIGPIO_ADDR/PIGPIO_PORT?
Feb 19 22:01:18 hassbian hass[583]: E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888
Feb 19 22:01:18 hassbian hass[583]: Did you specify the correct Pi host/port in the
Feb 19 22:01:18 hassbian hass[583]: pigpio.pi() function? E.g. pigpio.pi('soft', 8888)
Feb 19 22:01:18 hassbian hass[583]: %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
```

guymcswain, 2018-02-20:
It would be helpful if you can tell us how you initialized the library and what command you are executing when you get this error message.

EXTRAMINER, 2018-02-20:
The command is status home-assistant@homeassistant.service

joan2937, 2018-02-20:
Does homeassistant use pigpio?  Or are they both attempting to use the same socket (address 8888).

Perhaps tell one of them to use a different socket, e.g. `sudp pigpiod -p 8889` and the corresponding change to anything which uses pigpio.

EXTRAMINER, 2018-02-20:
@joan2937 yes,home assistant use pigpio but what is the second service that user them?

joan2937, 2018-02-20:
Is the pigpio daemon running, e.g. does the command `pigs t` work?  Can you start the daemon, e.g. `sudo pigpiod`?

EXTRAMINER, 2018-02-20:
I type `pigs t` and i get socket connect failed, then i typed `sudo pigpiod` and worked then i tried to run `pigs t` and worked with this output `1091308846`

joan2937, 2018-02-20:
That suggests that the problem was that the daemon hadn't been started.

You need to find a way of starting the daemon at boot which is suitable for your operating system.  Raspbian has a service so something like `sudo systemctl start pigpiod` works.

EXTRAMINER, 2018-02-20:
i think that the services is already booot startup. i typed `systemctl status pigpiod.service` and i get this:
```
● pigpiod.service - Daemon required to control GPIO pins via pigpio
   Loaded: loaded (/lib/systemd/system/pigpiod.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2018-02-20 13:51:41 UTC; 2min 53s ago
  Process: 20984 ExecStop=/bin/systemctl kill pigpiod (code=killed, signal=TERM)
  Process: 20981 ExecStart=/usr/bin/pigpiod -l (code=exited, status=0/SUCCESS)
 Main PID: 20982 (code=exited, status=1/FAILURE)

Feb 20 13:51:41 hassbian systemd[1]: Starting Daemon required to control GPIO pins via pigpio...
Feb 20 13:51:41 hassbian systemd[1]: Started Daemon required to control GPIO pins via pigpio.
Feb 20 13:51:41 hassbian pigpiod[20981]: 2018-02-20 13:51:41 initInitialise: Can't lock /var/run/pigpio.pid
Feb 20 13:51:41 hassbian pigpiod[20981]: Can't initialise pigpio library
Feb 20 13:51:41 hassbian systemd[1]: pigpiod.service: Main process exited, code=exited, status=1/FAILURE
Feb 20 13:51:41 hassbian systemd[1]: pigpiod.service: Unit entered failed state.
Feb 20 13:51:41 hassbian systemd[1]: pigpiod.service: Failed with result 'exit-code'.
```
i think this is the problem, how can i fix? thank you very much, also for my bad english

SlySven, 2018-02-20:
🤔  That looks like the previous daemon crashed and left behind a pidfile which is used to prevent more than one instance of the daemon running at a time - though I do not know whether the new fangled `systemd` stuff handles this cleanly {I cannot tell - I keep that daemon out of my systems and I wouldn't touch it with a tool used to propel a boat manually through the water, so my knowledge is getting to be a little out of touch these days - hell, I'd suggest taking a look through the system log files but even that is not as simple as it once was, in my, increasingly irrelevant, opinion... 🤷‍♂️ }

Have a look for a file `/run/pigpio.pid` and if it exists see if the number it contains matches up to a running process which could be providing the `pigiod` functionality - you will probably find it is there but no such process is running - then all you have to do is remove that file and try to run the daemon again...

EXTRAMINER, 2018-02-20:
I looked in `/run` and there isnt `pigpio.pid` file. Where can he is? It's a strange problem

joan2937, 2018-02-20:
As the message says it is in `/var/run/pigpio.pid`

EXTRAMINER, 2018-02-20:
Ah thanks. But the file, doesn't figure out. Why?

joan2937, 2018-02-20:
Sorry, I don't understand what you mean.

EXTRAMINER, 2018-02-20:
Sorry,i mean this. [(https://imgur.com/a/kWInM)]

joan2937, 2018-02-20:
The file should be present when the pigpio daemon is running.  It should be automatically deleted when the pigpio daemon stops.

If the pigpio daemon crashes the file may still be present and will stop a new version of the pigpio daemon from running.

If `pigs t` works the pigpio daemon is running okay.

If `pigs t` doesn't work then you should delete the file `/var/run/pigpio.pid` if it exists, e.g. `sudo rm /var/run/pigpio.pid`.

You can then try starting the pigpio daemon with `sudo pigpiod`.

EXTRAMINER, 2018-02-20:
When the raspberry start, if i type `pigs t` it give me `socket connect failed`, and from homeassistant pigpio doesn't work.
In `/var/run` there isnt `pigpio.pid`. If i put `sudo pigpiod` it works, also`pigs t` but from homeassistant it doesnt work, and the output of `sudo systemctl status pigpiod.service` is 
```
● pigpiod.service - Daemon required to control GPIO pins via pigpio
   Loaded: loaded (/lib/systemd/system/pigpiod.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
```

EXTRAMINER, 2018-02-20:
ok, thank you, now it works, i dont know why

EXTRAMINER, 2018-02-20:
It seems that the daemon doesnt start at the boot,or crash every time, how can i add to startup?

joan2937, 2018-02-20:
I have no idea.  I know nothing of hassbian.  I'm afraid you need to get the answer from them.

EXTRAMINER, 2018-02-20:
Umh, ok. if i found a solution, i will post them

SlySven, 2018-02-20:
> As the message says it is in `/var/run/pigpio.pid`

Ah, my `/var/run` is a symbolic link to `/run`... 😊

EXTRAMINER, 2018-02-20:
Now I fixed with a crontab, with inside
`@reboot /home/pi/pigpiod.sh`
And inside the script this command
```
#!/bin/bash
sudo systemctl stop pigpiod.service
sudo pigpiod
sudo systemctl stop home-assistant@homeassistant.service
sudo systemctl start home-assistant@homeassistant.service
echo Work done
```

PapaSquat, 2018-04-20:
I believe I'm having the same issue but the odd thing is the code only works great the first time after the pi is rebooted.  After that, it fails if I try to run it two or more times.  And there are two ways that it fails:
1. after the first run the callback from "e = pi.event_callback(pigpio.EVENT_BSC, i2c)" is never made. (when messages are received from the master).  Usually this error happens several times before the next one.
OR 
2. the following error will be produced.
```
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio at localhost(8888)

Did you start the pigpio daemon? E.g. sudo pigpiod

Did you specify the correct Pi host/port in the environment
variables PIGPIO_ADDR/PIGPIO_PORT?
E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888

Did you specify the correct Pi host/port in the
pigpio.pi() function? E.g. pigpio.pi('soft', 8888)
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
```

Either way, I have to reboot the pi in order to run the code again.

Sorry for the long message, I was trying to be thorough and not waste anyone's time. :)  I appreciate any help or ideas you may have!

Here's what I'm doing:
I'm trying to setup my pi as an I2C slave and I'm using the bsc_arduino.py example from [abyz.me.uk/...]
(http://abyz.me.uk/rpi/pigpio/examples.html#Python%20code).  The only changes I made to the example code was the I2C address, the sleep time, and I added a couple print statements.  I also added data to the method pi.bsc_i2c(addr,[data]).  My code below, at the end.

I have a feeling I'm making some silly mistake in the code because whenever I run the program after the first time, the status `s` returned from  `s, b, d = pi.bsc_i2c(I2C_ADDR)`,  is different.  In particular the bits (6 through 10) representing the number of bytes in the transmit FIFO always seems to start at a different, non-zero value.  And if you re-run the program back-to-back that value does not change sequentially.

After a fresh reboot and before running the program for the first time this is the pigpiod process info I get from the terminal
```
pi@raspberrypi:~ $ ps aux | grep pigpiod
root       749  7.0  0.1  11440  1676 ?        SLsl 17:17   1:15 pigpiod
pi         950  0.0  0.0   4376   580 pts/0    S+   17:35   0:00 grep --color=auto pigpiod
```
Once I get the `Can't connect to pigpio at localhost(8888)` error the entry for 'root' disappears but the entry for pi remains.  And the file /var/run/pigpio.pid is always there.  Deleting that file and relaunching pigpiod (sudo pigpiod) still does not fix the issue.  However, it does upgrade me to failure #1 described at the top, rather than #2.

Other Details:
- I have a Raspberry Pi 3b+, running Raspbian Stretch (v9).
- Using Python v3.5.3
- I added `i2c_bcm2708` to the /etc/modules file after reading the suggestion in an Adafruit guide [here](https://cdn-learn.adafruit.com/downloads/pdf/adafruits-raspberry-pi-lesson-4-gpio-setup.pdf) (section "Installing Kernal Support (Manually)" page 17).  It seems to help but I have no idea if that's the right thing to do or not.
- This also could be related to issue #203 but I'm not sure since I start pigpiod at startup by:
creating the file ~/bin/startup_script, adding necessary commands, give it execute permissions, and calling that file by adding it to the end of /home/pi/.config/lxsession/LXDE-pi/autostart

Here's the code I'm using.
```
#!/usr/bin/env python

# bsc_arduino.py
# 2016-10-31
# Public Domain

"""
This code is an example of using the Pi as an I2C slave device
receiving messages from an Arduino acting as the I2C bus master.

SKIPPING THE REMAINING COMMENTS
"""

import time
import pigpio

##pigpio.exceptions = False

SDA=18
SCL=19

I2C_ADDR=0xB4

def i2c(id, tick):
   global pi
   
   data = [0x01,0x02,0x03,0x04,0x05]
   s, b, d = pi.bsc_i2c(I2C_ADDR,data)
   
   
   #DEBUGGING CODE
   s = format(s,'021b')
   print('s = {}'.format(s))
   print('b = {}'.format(b))
   print('d = {}'.format(d))
   ###############
   
   
   if b:

      print(d[:-1])

pi = pigpio.pi()

if not pi.connected:
    print('pi.connected = False')
    exit()

# Add pull-ups in case external pull-ups haven't been added

pi.set_pull_up_down(SDA, pigpio.PUD_UP)
pi.set_pull_up_down(SCL, pigpio.PUD_UP)

# Respond to BSC slave activity

pigpio.exceptions = False

e = pi.event_callback(pigpio.EVENT_BSC, i2c)

s, b, d = pi.bsc_i2c(I2C_ADDR) # Configure BSC as I2C slave

#DEBUGGING CODE
print('\nJust Executed pi.bsc_i2c(I2C_ADDR)')
s = format(s,'021b')
print('s = {}'.format(s))
print('b = {}'.format(b))
print('d = {}'.format(d))
###############

time.sleep(1)

e.cancel()

pi.bsc_i2c(0) # Disable BSC peripheral

pi.stop()

```

joan2937, 2018-05-11:
Perhaps something else is using that port?

All versions are intended to be available at http://abyz.me.uk/rpi/pigpio/VERSIONS/

indiependente, 2018-10-01:
V67 failing for me as well on any port
```
[root@alarm ~]# pigpiod -p 8888
2018-10-01 01:31:27 initPeripherals: mmap gpio failed (Operation not permitted)
[root@alarm ~]# Can't initialise pigpio library

[root@alarm ~]# pigpiod -p 8889
2018-10-01 01:31:29 initPeripherals: mmap gpio failed (Operation not permitted)
Can't initialise pigpio library
```

Here is a strace of an execution tentative: https://hastebin.com/hezelayuba.cpp

indiependente, 2018-10-01:
@guymcswain how can I track down which is the other library and eventually remove it?
Did you notice that from the strace? 
Thanks.

guymcswain, 2018-10-02:
> Did you notice that from the strace?

No I glanced at the trace but really don't understand it well enough to comment.   I'm going by the error message "**mmap gpio failed (Operation not permitted)**"

I don't have a good idea on how to discern if another process has already mapped the peripheral.  I'm assuming there would be a lock on the ability to map it.  Sorry I can't be more helpful.

Perhaps you should open a new issue?

chasihler, 2018-12-24:
I had a similar issue. I had to edit my start up service file because by default the newer pi binds to ip6 over ip4.
This fix my issue:

/lib/systemd/system/pigpiod.service

[Service]
ExecStart=/usr/bin/pigpiod -l -n 127.0.0.1

davidwachs, 2019-01-24:
This fixed it for me!  Thank you chasihler!

guymcswain, 2020-01-03:
Duplicate of #203.

guymcswain, 2020-01-04:
@PapaSquat ,
I believe the issue you describe above is different and not related to inability to connect using 'localhost'/127.0.0.1 IPv4.  Can you please do the following:

1. To ensure you are **not** experiencing the IPv4 problem please run the daemon (pigpiod) without the `-l` option.
2. If you still see the anomalous behavior, open a new issue as I believe you may be experiencing an issue with an I2C API or other.

guymcswain, 2020-01-10:
A work-around for this issue is posted on the wiki [here](https://github.com/joan2937/pigpio/wiki).

guymcswain, 2020-01-14:
Confirmed this issue is resolved after update to pigpio V73.

closed 2020-01-14 by guymcswain

andreapx, 2023-04-02:
I've just installed V79 and I have this problem, pigpiod binds to ::1:8888 and telnet 127.0.0.1 8888 fails.
I've edited /lib/systemd/system/pigpiod.service as posted up here:

```[Unit]
Description=Daemon required to control GPIO pins via pigpio
[Service]
ExecStart=/usr/bin/pigpiod -l -n 127.0.0.1
ExecStop=/bin/systemctl kill pigpiod
Type=forking
[Install]
WantedBy=multi-user.target
```

But the problem persists.