joan2937/pigpio #196: *_EDGE not detecting first event

arcsyst, 2018-02-20:
Forgive me if this is a stupid question but I am migrating from the older rpi.GPIO based event handling. Using your monitor.py to test my understanding of event handling, and gpio_status.py as a port status viewer at the same time, I can see the port transition from 0 to 1 (I'm detecting RISING_EDGE) but the event isn't triggered, if I transition 1 to 0 then 0 to 1 again it fires on the second (and subsequent) transitions 0-1. Is there a way to catch the first event?

joan2937, 2018-02-20:
You can modify the behavior by changing

```python
def cbf(GPIO, level, tick):
   if last[GPIO] is not None:
      diff = pigpio.tickDiff(last[GPIO], tick)
      print("G={} l={} d={}".format(GPIO, level, diff))
   last[GPIO] = tick
```
to
```python
def cbf(GPIO, level, tick):
   if last[GPIO] is not None:
      diff = pigpio.tickDiff(last[GPIO], tick)
      print("G={} l={} d={}".format(GPIO, level, diff))
   else:
      print("G={} l={}".format(GPIO, level))
   last[GPIO] = tick
```
The first time through you don't have any level duration information.

arcsyst, 2018-02-20:
Thanks for the prompt reply, it never occurred to me that the problem was in the called code I just assumed it was in the pigpio code. My bad.

Thanks for an excellent bit of code, my next task is to get my lightwave RF devices to work through it as the old code seems to not work on a Pi3.

Thanks again
Brian

guymcswain, 2020-01-02:
No follow up action required.  Closing.

closed 2020-01-02 by guymcswain