joan2937/pigpio #197: faulty (?) RFID-Readings with python pigpio

sk-gara, 2018-02-21:
I'm reading RFID cards from Honeywell-Reader on a Pi3. Taking the wiegand.py example from the homepage as a start, i modified the data- and clock pin watchdogs (only on clock) and the callback functions. Works well. By the way, from inside one or more docker containers or directly on the host OS, which is Raspbian.
But when stress-testing the function with many card-scans in succesion, after some correct readings, there will follow up some corrupted values - caused by tick-values coming from the clock pin that are too "long" or too "short", triggering the watchdog on that pin too early or too late.
When monitoring the readings with Piscope in parallel, i can see that the readings are correct, so the tick-values presented by the python package seem to be somehow corrupted.
Is this a known limitation of the python implementation of pigpio or should it work as flawless as it runs in Piscope?

I'm using piscope 64.

Here some values to illustrate, the columns are:
 tick, ms since last tick, pin, data_level
```
547086377 547086377 17 1
547087582 1205 17 1
547088797 1215 17 1
547089937 1140 17 1
547091122 1185 17 1
547092268 1146 17 1
547093412 1144 17 1
547094553 1141 17 1
547095772 1219 17 1
547097007 1235 17 1
547098188 1181 17 1
547099329 1141 17 1
547100509 1180 17 1
547101653 1144 17 1
547102863 1210 17 1
547104003 1140 17 1
547105147 1144 17 1
547106287 1140 17 1
547107612 1325 17 1
547108707 1095 17 1
547109853 1146 17 1
547111044 1191 17 1
547118538 7494 17 2
```

joan2937, 2018-02-21:
Can you provide a self contained script which demonstrates the problem.

Both piscope and the Python module use the same mechanism to receive GPIO events.  Of course the Python module handles them differently, but I'd expect the same events to be received in the same order.

joan2937, 2018-02-22:
I ran [monitor.py](http://abyz.me.uk/rpi/pigpio/examples.html#Python_monitor_py) with output to a log file.

I then used the following commands.

`pigs pfs 4 50 pfs 5 100 pfs 6 200 pfs 7 400 pfs 8 800`

Set frequencies 50, 100, 200, 400, 800 Hz.

`pigs p 4 40 p 5 50 p 6 60 p 7 70 p 8 80`

To send PWM at various dutycycyles.

I left the monitor running until the log was about 30 megs.

I then discarded the first and last hundred entries so as not to have any start up or end anomalies.  Sort'd and uniq'd gave the following.  No obvious problem.

```
    945 G=4 l=0 d=3134
  34969 G=4 l=0 d=3135
    793 G=4 l=0 d=3136
    301 G=4 l=1 d=16864
  34095 G=4 l=1 d=16865
   2311 G=4 l=1 d=16866
   1885 G=5 l=0 d=1959
  69073 G=5 l=0 d=1960
   2457 G=5 l=0 d=1961
   1644 G=5 l=1 d=8039
  68841 G=5 l=1 d=8040
   2930 G=5 l=1 d=8041
   3707 G=6 l=0 d=1174
 138798 G=6 l=0 d=1175
   4324 G=6 l=0 d=1176
   3610 G=6 l=1 d=3824
 138369 G=6 l=1 d=3825
   4851 G=6 l=1 d=3826
   8262 G=7 l=0 d=684
 277491 G=7 l=0 d=685
   7907 G=7 l=0 d=686
   6427 G=7 l=1 d=1814
 278593 G=7 l=1 d=1815
   8640 G=7 l=1 d=1816
  11967 G=8 l=0 d=389
 563483 G=8 l=0 d=390
  11869 G=8 l=0 d=391
  14611 G=8 l=1 d=859
 556141 G=8 l=1 d=860
  16567 G=8 l=1 d=861

```

sk-gara, 2018-02-22:
Here's the code i use for testing, similar to the code working in the program:

```python
import pigpio
import time
import logging
from logzero import logger
from logzero import logfile

class Reader:
    def __init__(self):

        self.watchdog_trailing_ms = 6
        self.pin0 = 27
        self.pin1 = 17

        logfile('pigpio_readings.log')
        self.logger = logger

    def decode(self):

        self.pi = pigpio.pi()
        self.is_reading = True
        self.in_code = False
        self.tick_dict = {}

        self.pi.set_mode(self.pin0, pigpio.INPUT)
        self.pi.set_mode(self.pin1, pigpio.INPUT)

        self.pi.set_pull_up_down(self.pin0, pigpio.PUD_UP)
        self.pi.set_pull_up_down(self.pin1, pigpio.PUD_UP)

        self.cb_0 = self.pi.callback(self.pin0, pigpio.EITHER_EDGE, self._cbf)  # Data
        self.cb_1 = self.pi.callback(self.pin1, pigpio.RISING_EDGE, self._cbf)  # Clock

    def _cbf(self, gpio, level, tick):

        if level < pigpio.TIMEOUT:

            if not self.in_code:
                """ starting, reset all variabels """

                self.data_level = 0
                self.bits = 0
                self.in_code = True
                self.tick_dict = {}
                self.tick_count = 0
                self.pi.set_watchdog(self.pin1, self.watchdog_trailing_ms)

            else:

                self.tick_dict[self.tick_count] = [tick, gpio, level]
                self.tick_count += 1

                if gpio == self.pin0:
                    self.data_level = level

                else:  # clock rising edge
                    self.bits = int(self.bits)
                    self.bits = "{}{}".format(self.bits, self.data_level)

        else:
            self.tick_dict[self.tick_count] = [tick, gpio, level]

            self.logger.debug('clock level time_out')
            time.sleep(self.sleep_time)

            if self.in_code:

                self.pi.set_watchdog(self.pin1, 0)
                self.in_code = False

                value = self.make_value(self.bits, self.logger, self.sleep_time, self.tick_dict)

                self.tick_count = 0
                self.tick_dict = {}

    @staticmethod
    def make_value(value_bin, logger, sleep_time, tick_dict):

        # print(value_bin)
        if len(str(value_bin)) == 115:
            logger.debug(value_bin)
            # calculate value from bin...

        else:  # print warning
            last_tick = 0
            clock_ticks = {}
            logger.warn(value_bin)

            for item in tick_dict.items():
                value = item[1]
                tick = value[0]
                pin = value[1]

                if pin == 17:
                    level = value[2]
                    warn_string = '{} {} {} {}'.format(tick, tick - last_tick, pin, level)
                    logger.warn(warn_string)
                    last_tick = tick

    def cancel(self):
        if self.cb_0:
            self.cb_0.cancel()
        if self.cb_1:
            self.cb_1.cancel()
        if self.pi is not None:
            self.pi.stop()

if __name__ == '__main__':
    reader = Reader()

    while True:
         reader.decode()
         time.sleep(30)
```

And I'm gonna try monitor.py with the reader.

joan2937, 2018-02-22:
The code has dependencies on a module called logzero which I don't have.

Which Python version are you using?

sk-gara, 2018-02-22:
logzero is just a convenient wrapper around the logging module. I replaced it with a simple print() and it made no difference.
But what actually seemed to fix the problems with the watchdog-timeouts is how you call the infinite loop on the Reader. In my example the decode() function gets called every 30 seconds again, whereas when i initiate the Reader and set the callbacks in __init__() and after that do the infinite time.sleep - that _seems_ to do the trick. I have to do more testing tomorrow and will let you know if that really was it.

Python 3.5.3 is it, by the way.
---

when quick testing on another Pi3 (Python 3.5.3 also), i got wrong result again. So, possibly something regarding the system setup is interrupting the process. Or so. Need to look further into that tomorrow.

joan2937, 2018-03-02:
Have you got anywhere with this?  If you still feel there is an error could you prepare a script which demonstrates the problem (only using standard installed modules).

sk-gara, 2018-03-02:
First, sorry for the delay! There were other issues i had to attend to. Thanks for your patience.
I isolated a script that shows how the Reader is running in my app:
```
import pigpio
import time

class Reader:
    def __init__(self):

        self.watchdog_trailing_ms = 6

        self.pin0 = 27
        self.pin1 = 17

        self.read_count = 0

        print('initialize GPIOs...')

        self.pi = pigpio.pi()

        self.is_reading = True

        self.in_code = False

        self.pi.set_mode(self.pin0, pigpio.INPUT)
        self.pi.set_mode(self.pin1, pigpio.INPUT)

        self.pi.set_pull_up_down(self.pin0, pigpio.PUD_UP)
        self.pi.set_pull_up_down(self.pin1, pigpio.PUD_UP)

        self.pi.set_watchdog(self.pin1, 0)

        self.cb_0 = self.pi.callback(self.pin0, pigpio.EITHER_EDGE, self._cbf)  # Data
        self.cb_1 = self.pi.callback(self.pin1, pigpio.RISING_EDGE, self._cbf)  # Clock

    def _cbf(self, gpio, level, tick):

        """

        :param gpio:
        :param level:
        :param tick: The number of microseconds since boot
        :return:
        """

        if level < pigpio.TIMEOUT:

            if not self.in_code:
                """ starting, reset all variabels """

                self.data_level = 0
                self.bits = 0
                self.in_code = True
                self.code_timeout = 0

                self.pi.set_watchdog(self.pin1, self.watchdog_trailing_ms)

            else:

                if gpio == self.pin0:
                    self.data_level = level

                else:  # clock rising edge
                    self.bits = int(self.bits)
                    self.bits = "{}{}".format(self.bits, self.data_level)
                    self.code_timeout = 0

        else:

            if self.in_code:
[... 132 lines cut ...]
but eventually, the values get out of order:
```
# 19
1101000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111 code_error: value not allowed: 15
# 20
invalid bit_value: 00
# 21
invalid bit_value: 00
# 22
invalid bit_value: 00
# 23
invalid bit_value: 00
# 24
invalid bit_value: 00
# 25
invalid bit_value: 00
# 26
invalid bit_value: 00
# 27
invalid bit_value: 00
# 28
invalid bit_value: 0
# 29
1101000001110011000000100000010000100001000011000010000000010000110000000011000011001100000010000001010001111101000 03140000110010131402
```

joan2937, 2018-03-02:
Is there any simple way to generate test data?  If I can feed data in I can leave it running and perhaps get a feel for what may be going wrong.

sk-gara, 2018-03-05:
I will try and write a script with artificial signal generation. I'm curious if the result will be same as with a hardware-reader.

sk-gara, 2018-03-06:
here's a simple script, that emulates a 20-ticks clock signal:

```
import pigpio
import time

class SignalGenerator:
    def __init__(self):

        self.pin1 = 17  # clock
        self.pi = pigpio.pi()
        self.pi.set_mode(self.pin1, pigpio.OUTPUT)
        self.pi.set_pull_up_down(self.pin1, pigpio.PUD_UP)

        while True:
            try:
                for i in range(0, 19):
                    self.pi.write(self.pin1, 1)
                    time.sleep(.0012)
                    self.pi.write(self.pin1, 0)
                    time.sleep(.0012)

                time.sleep(1)

            except KeyboardInterrupt:
                print("\nTidying up")
                self.cancel()

    def cancel(self):

        if self.pi is not None:
            self.pi.stop()

gen = SignalGenerator()
```

then i observe pin 17 outputs with a modified `monitor.py`:
```
import sys
import time
import pigpio

last = [None] * 32
cb = []

def cbf(GPIO, level, tick):

    pi.set_watchdog(g, 4)

    if last[GPIO] is not None:
        diff = pigpio.tickDiff(last[GPIO], tick)
        print("GPIO: {} level: {} diff: {}".format(GPIO, level, diff))

    last[GPIO] = tick

    if level == pigpio.TIMEOUT:
        print('  timeout')
        pi.set_watchdog(g, 0)

pi = pigpio.pi()

if not pi.connected:
    exit()

if len(sys.argv) == 1:
    G = range(0, 32)
else:
    G = []
for a in sys.argv[1:]:
    G.append(int(a))

for g in G:
    cb.append(pi.callback(g, pigpio.RISING_EDGE, cbf))

try:
    while True:
        time.sleep(60)
except KeyboardInterrupt:
    print("\nTidying up")

    for c in cb:
        c.cancel()

pi.stop()
```

after a short while, there will be blocks of less than 20 ticks:

GPIO: 17 level: 1 diff: 998423
GPIO: 17 level: 1 diff: 3085
GPIO: 17 level: 1 diff: 3070
GPIO: 17 level: 1 diff: 3010
GPIO: 17 level: 1 diff: 3015
GPIO: 17 level: 1 diff: 2995
GPIO: 17 level: 1 diff: 2985
GPIO: 17 level: 1 diff: 3005
GPIO: 17 level: 1 diff: 2990
GPIO: 17 level: 1 diff: 3070
GPIO: 17 level: 1 diff: 2990
GPIO: 17 level: 1 diff: 2995
GPIO: 17 level: 1 diff: 2970
GPIO: 17 level: 1 diff: 3000
GPIO: 17 level: 1 diff: 3080
GPIO: 17 level: 1 diff: 2995
GPIO: 17 level: 1 diff: 3005
GPIO: 17 level: 1 diff: 2995
GPIO: 17 level: 1 diff: 2980
GPIO: 17 level: 2 diff: 6345
  timeout
GPIO: 17 level: 1 diff: 997763
GPIO: 17 level: 1 diff: 2839
GPIO: 17 level: 1 diff: 2781
GPIO: 17 level: 1 diff: 2765
GPIO: 17 level: 2 diff: 5479     # too early
  timeout

joan2937, 2018-03-06:
Thanks for that.  I'll add it to the list of things to be looked at.  It's number 4 in the queue though so don't expect a quick response.

joan2937, 2018-03-13:
The monitor code you posted is setting a watchdog on GPIO 31.  Is global variable g meant to be parameter GPIO?

`    pi.set_watchdog(g, 4)
`

joan2937, 2018-03-13:
I can find no evidence of a problem.  If I was to guess I suggest the watchdog timeout is too short.

You seem to be starting a watchdog for each bit.  That might be the right thing to do but could you not set a watchdog for the complete message?  I.e. check if bit 1 has just been received and then set a watchdog for 20 bit times plus an allowance?  That would be one watchdog per message rather than 20.

joan2937, 2018-03-14:
I have fed over a hundred thousand messages into your code.
```
pi3b:~ $ pigs wvag $((1<<17)) 0 500 0 $((1<<17)) 500
2
pi3b:~ $ pigs wvcre
1
pi3b:~ $ pigs wvcha 255 0 255 0 1 255 1 20 0 255 2 0x20 0x4e 255 3
pi3b:~ $ 
```
 with the modified callback as follows.
```c
def cbf(GPIO, level, tick):

    global bits

    bits += 1

    if bits == 1:
       pi.set_watchdog(GPIO, 5)

    if last[GPIO] is not None:
        diff = pigpio.tickDiff(last[GPIO], tick)
        #print("GPIO: {} level: {} diff: {}".format(GPIO, level, diff))

    last[GPIO] = tick

    if level == pigpio.TIMEOUT:
        print('  timeout, bits={}'.format(bits))
        pi.set_watchdog(GPIO, 0)
        bits = 0
```
There were no errors.

```
pi3b:/ram $ sort log|uniq -c
 108001   timeout, bits=21
pi3b:/ram $
```

sk-gara, 2018-03-15:
Thanks for investigating. When i test with pigs, i get no errors as well.

And you are right about the watchdog. But in the the 'real world' class i'm using, it's one watchdog per message, as it should be. Also, it makes no difference for me if i change it in the test script.
I think I will need to test further with different environments.
Do you have any hardware RFID reader that you could use for testing?

sk-gara, 2018-07-23:
Hey Joan,

I'm sorry for having abandoned this issue for a while. We approached it with catching and discarding read-errors at software-level by evaluating the read code. Still occurs but doesn't really limit functionality now.
Anyway, thanks for your time and for this library!

closed 2018-07-23 by sk-gara