joan2937/pigpio #199: git tag "releases"

kripton, 2018-02-24:
Hi, thank you so much for that library!
This issue is not about a problem in the code but in the "release process". On the download page of the library (http://abyz.me.uk/rpi/pigpio/download.html) the current "release" is described as V66 but the download instructions use the "master.zip" from github or an unversioned archive from the homepage in any case.
Would you mind creating (and pushing) a git tag if you "release a new version"? (for example tag "V66" on commit c734018f3c98ef979d8cc9f7878415894843cb48. This way, GitHub will automatically have an URL to download that specific release and it would make it easier to refer to a specific version of even integrate the library into linux distributions.
Thank you :)

kripton, 2018-02-24:
Creating git tags could also be done retroactive on already existing release commits

SlySven, 2018-02-24:
If you know the SHA1 you can get a specific tarball of the source code with:

`https://codeload.github.com/`\<repository owner>`/`\<repository>`/tar.zg/`\<SHA1>

so for that commit you'd need:

[`https://codeload.github.com/joan2937/pigpio/tar.gz/c734018f3c98ef979d8cc9f7878415894843cb48`](https://codeload.github.com/joan2937/pigpio/tar.gz/c734018f3c98ef979d8cc9f7878415894843cb48)

the same thing also works for the `.zip` archive form if you substitute `zip` for `tar.gz` in the above.

If @joan2937 does tag the releases in a particular way (I am not completely sure how, but see: [About Releases](https://help.github.com/articles/about-releases) and [Creating Releases](https://help.github.com/articles/creating-releases) for clues) - then there is a shorter URL way to get those!

kripton, 2018-02-24:
Indeed, one can download every commit as archive from GitHub. The problem is that the SHA1 hashes are not sortable so you don't know which commit is newer or older just by looking at it. With version tags this is easy.
Of course one could keep a mapping of versions so a linux distribution knows which SHA1 hash corresponds to a specific version but since GitHub offers that feature as well by just using tags that would be the way to go

joan2937, 2018-02-25:
I have generally no objection in doing things to make life easier.

However I am not familiar with GIT, and do not plan to spend time getting familiar.

Do you have clear instructions as to how I achieve what you want?

If not much is required I will almost certainly do it.  If it requires significant effort I probably will not.

kripton, 2018-02-25:
That sounds good. The process is fairly easy, it's just two additional steps. Assuming you you the git command line, after having created the release commit (the ones with only the version number as commit message, as in "V66"), you run ```git tag V66```. That creates the tag. In addition to the usual ```git push``` you will have to run ```git push --tags``` so that the tags are published as well.
The rest will automatically be handled by GitHub.
If you want to retroactively tag a commit, you check it out (switch the git index to that commit) and run the two steps.
Thank you!

kripton, 2018-02-25:
Perfect, https://github.com/joan2937/pigpio/archive/V66.tar.gz is working. Thank you :+1:

closed 2018-02-25 by kripton

SlySven, 2018-02-26:
FTR - it is possible to extend the rather automagically generated sparse details on the "Releases" detail:
[image]

This does give the bare minimum: 
[image]

But for a repository owner (or those with the relevant permission) it is a useful place to report the latest changes from a ChangeLog, e.g.:
[image]