joan2937/pigpio #201: pigpiod stops working after stack smashing

grillbaer, 2018-02-26:
Hi,

Thank you very much for your great tool pigpio. I really like it! However, I ran into an issue and hopefully somebody can help me.

In my application pigpiod regularly stops working after some time. The pigpiod process then is still alive, but one of its threads died after gcc's checking found a corrupted stack. This then requires killing and restarting pigpiod and my app:

```
pi@raspi3:~ $ cat /dev/pigerr
*** stack smashing detected ***: /usr/bin/pigpiod terminated
2018-02-24 15:20:18 sigHandler: Unhandled signal 6, terminating
```

After removing the signal handler for `SIGABRT 6` from the code and adding debug options in the Makefile (`-g -O0`, removing the strip commands) I was able to create a core dump. It shows an overwrite beyond the limit of array `report[MAX_REPORT]` in function `alertEmit()`:

```
root@raspi3:~/PIGPIO/crash# gdb pigpiod core
GNU gdb (Raspbian 7.12-6) 7.12.0.20161007-git
...
Reading symbols from pigpiod...done.
[New LWP 430]
[New LWP 541]
[New LWP 542]
[New LWP 431]
[New LWP 434]
[New LWP 428]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Core was generated by `pigpiod'.
Program terminated with signal SIGABRT, Aborted.
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51   ../sysdeps/unix/sysv/linux/raise.c: Datei oder Verzeichnis nicht gefunden.
[Current thread is 1 (Thread 0x767b7470 (LWP 430))]
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x76cdc824 in __GI_abort () at abort.c:89
#2  0x76d15f78 in __libc_message (do_abort=do_abort@entry=2, fmt=<optimized out>) at ../sysdeps/posix/libc_fatal.c:175
#3  0x76d92264 in __GI___fortify_fail (msg=0x76dd1d90 "stack smashing detected") at fortify_fail.c:30
#4  0x76d9221c in __stack_chk_fail () at stack_chk_fail.c:28
#5  0x76e47a20 in alertEmit (sample=0x767af164, numSamples=120, changedBits=3211264, eTick=1108643017) at pigpio.c:6128
#6  0x243991fe in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) fr 5
#5  0x76e47a20 in alertEmit (sample=0x767af164, numSamples=120, changedBits=3211264, eTick=1108643017) at pigpio.c:6128
...
(gdb) p report[0]
$18 = {seqno = 21513, flags = 0, tick = 1108637207, level = 606704111}
(gdb) p report[1]
$19 = {seqno = 21514, flags = 0, tick = 1108637267, level = 607752687}
(gdb) p report[119]
$20 = {seqno = 21632, flags = 0, tick = 1108643017, level = 607752703}
(gdb) p report[120]
$21 = {seqno = 21633, flags = 44, tick = 1108643017, level = 607752703}
(gdb) p report[121]
$22 = {seqno = 21634, flags = 48, tick = 1108643017, level = 607752703}
(gdb) p report[122]
$23 = {seqno = 21635, flags = 51, tick = 1108643017, level = 607752703}
(gdb) p report[123]
$25 = {seqno = 21636, flags = 52, tick = 1108643017, level = 607752703}
(gdb) p report[124]
$26 = {seqno = 21637, flags = 58, tick = 1108643017, level = 607752703}
(gdb) p report[125]
$27 = {seqno = 0, flags = 0, tick = 2126326868, level = 0}
(gdb) p report[126]
$28 = {seqno = 4096, flags = 1081, tick = 33675, level = 1108646412}
```

The array size is `MAX_REPORT=120`, it has been written until `report[124]` in this case.

My setup:

Raspberry PI 3 with up-to-date Raspbian (Jessie) and the latest PIGPIO V66. 
```
root@raspi3:~/PIGPIO# cat /etc/debian_version
9.3
root@raspi3:~/PIGPIO# uname -a
Linux raspi3 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux
root@raspi3:~/PIGPIO# pigpiod -v
66
```

I'm reading signals of 6 TSIC 306 temperature sensors connected to 6 GPIOs using callbacks for both signal edges in a Python 3 application. Each of the sensors sends two bytes 10 times a second with the ZacWire protocol. This is about 10*40 signal edges per sensor per second. The application also sets watchdog timers to detect the ends of transmissions.

The problem seems to depend on timing. My tests with stripped down applications also caused the crash, but by far not as frequently.

Please let me know how I can help to fix this issue.

Best regards
Holger

joan2937, 2018-02-26:
The most useful thing would be a script which triggers the problem.  Perhaps using PWM to generate edges similar to those being generated by the sensors.

grillbaer, 2018-02-27:
Ok, I'll have a look at how some stress test could be simulated.

Meanwhile, I tested increasing the buffer size `MAX_REPORT` from 120 to 1200 and pigpiod worked without any problems for over 20 hours now. With the 120 entry buffer it usually crashed after 10 to 30 minutes.
Of course that is not a clean solution.

joan2937, 2018-02-27:
Don't worry about the stress test unless it is easy.

Thanks for the very detailed report.

I looked in more detail at your core dump and I think it shows the problem.

If a watchdog or event is due for the notification I stupidly append the report to the array without checking that there is room.  It appears to be a straightforward bug.

I might simply dimension the array to be MAX_REPORT + 64 to allow for the maximum number of watchdogs and events (1 each per GPIO).  I chose MAX_REPORT to be 120 for a reason i can no longer remember, may have been ethernet frame size.

If you get a chance could you change

`gpioReport_t report[MAX_REPORT];`

to

`gpioReport_t report[MAX_REPORT + 64];`

in alertEmit()

I will add a fix for this bug in V67.

grillbaer, 2018-02-27:
This is easy... the test with `report[MAX_REPORT + 64]` is already running, now. I'll give some feedback tomorrow.

grillbaer, 2018-02-28:
The test with buffer `report[120 + 64]` has been running for 15 hours without problems. Looks good!

Thank you for your help!

joan2937, 2018-02-28:
Good, thanks.

In that case your core dump has probably identified the actual bug.

closed 2018-03-20 by joan2937