joan2937/pigpio #202: APDS9960: Clear Interrupt registers

Joey-1970, 2018-02-28:
Hello,

to clear the interrup register of a APDS9960 seemd to be a special command:

> Clear Interrupt Registers (0xE4 – 0xE7)
Interrupts are cleared by “address accessing” the appropriate register. This is special I2C transaction consisting of only two bytes: chip address with R/W = 0, followed by a register address.

Which PIGPIO-command should I use?
I try a I²C-write command, but it look like it don't work...

Joachim

Joey-1970, 2018-03-01:
...from the I²C-Specification:

> This address is seven bits long followed by an eighth bit which is a data
direction bit (R/W) — a ‘zero’ indicates a transmission (WRITE), a ‘one’ indicates a
request for data (READ)

From that point, I have to do a write-command. But I see one two others places of the same APDS9960-datasheet:

> cleared by “address accessing” (i.e. I²C transaction consisting of only two bytes: chip address, followed by a register address with R/W=1) PICLEAR or AICLEAR

I'm confused...

Joachim

joan2937, 2018-03-01:
This isn't really a pigpio issue.

You need to write one byte to the device (the register you want to clear).

e.g. [pigs i2cws](http://abyz.me.uk/rpi/pigpio/pigs.html#I2CWS), [Python i2c_write_byte](http://abyz.me.uk/rpi/pigpio/python.html#i2c_write_byte)

guymcswain, 2020-01-02:
No follow up action is required.  Closing.

closed 2020-01-02 by guymcswain