joan2937/pigpio #204: pigpiod reboots RPi3 when phat beat hat installed

secolaha, 2018-03-04:
Hi, 
If I start pigpiod on a RPI3 with a phat beat hat installed, the Pi crashes and reboots itself.

I suspect that there are pins that the phat beat are using that conflicts with how the pigiod is working.
is there a way to exclude certain pins in pigpio so that it wont us them?
here is the pinout of the phat beat hat.

regards, 
/Lars

joan2937, 2018-03-04:
pigpio does not use any GPIO until asked to do so.

It's more likely that the conflict is with the PWM or PCM peripheral.

By default pigpio uses the PCM peripheral.  Try starting pigpio with the use PWM peripheral option instead.

See http://abyz.me.uk/rpi/pigpio/pigpiod.html

secolaha, 2018-03-04:
Thank you for your quick reply!
I will try that.

regards, 
/Lars

guymcswain, 2020-01-02:
No follow up action required.  Closing.

closed 2020-01-02 by guymcswain