joan2937/pigpio #205: Installing pigpio via pip does not install pigpio.py

sphh, 2018-03-04:
I tried to install pigpio with the following command on a Raspberry Pi for Python 3.4:
```
sudo pip3 install pigpio
```
This tries to install pigpio 1.40 and only makes the directory `/usr/local/lib/python3.4/dist-packages/pigpio-1.40.dist-info`, but no file `/usr/local/lib/python3.4/dist-packages/pigpio.py`.

Installing the previous version 1.38 with
```
sudo pip3 install pigpio==1.38
```
works fine.

joan2937, 2018-03-04:
I haven't put anything under the pip system.  That is maintained by a third party who may update the version on an ad hoc basis.

closed 2018-03-04 by joan2937

sphh, 2018-03-04:
You are right. The smallprint tells me that it's @bennuttall. I already contacted him. Sorry, I should have checked!

bennuttall, 2018-03-06:
Sorry about this - I'm not sure how it happened but I'll push up a fixed version asap.

bennuttall, 2018-03-06:
Ok that's done. You can't replace files once they're uploaded, so I have uploaded a `1.40.post1` release.