joan2937/pigpio #206: gpioSetISRFunc relative high cpu usage

st42, 2018-03-05:
I recently found pigpio and I decided to try it because of the hardware timed pwm. I haven't tried pwm yet, but I have rewrote an old program using the pigpio C library to compare it with the wiringpi C library. I am using the following code:

```
#include <stdlib.h>
#include "pigpio.h"

#define PIN 21

void my_cb(int gpio, int level, uint32_t tick)
{
	if(level==0) system("sudo halt");
}

int main(void) {
	gpioInitialise();
	gpioSetMode(PIN, PI_INPUT);
	gpioSetPullUpDown(PIN, PI_PUD_UP);
	gpioSetISRFunc(PIN, EITHER_EDGE, 1000, my_cb);
	while(1) time_sleep(1000);
	gpioTerminate();
	return 0;
}
```
All the code does is shutdown the Pi when the pin is grounded. It is run at boot with `/etc/rc.local`
The code as is uses 8-9% cpu which isn't a lot but wiringpi with similar code doesn't even register cpu usage. This seems high for an interrupt.

My question is am I initializing the interrupt wrong or is the cpu usage just a byproduct of the way the hardware timing is initialized? I did notice that if I start `pigpiod` without running any code, it uses the same amount of cpu (8-9%).

joan2937, 2018-03-05:
That is the base load.  It is generated by the sampling of the GPIO at 200 thousand times per second.  More accurately the load is generated by the code which wakes a thousand times per second to see which GPIO have changed state and would alert any process which has registered an interest.

See http://abyz.me.uk/rpi/pigpio/cif.html#gpioCfgClock for timings I did years ago.

st42, 2018-03-05:
I suspected that was the case. Thanks for the quick response.

closed 2018-03-05 by st42