joan2937/pigpio #209: Is it possible to create a serial `/dev/tty*` style device from the GPIO interface?

ShakenU, 2018-04-16:
So, my team is using a CM3, and we need a minimum of 3 USART lines. Two of the devices run at a very high baud and will be sending large amounts of data, however, the third one will be infrequently used and need, at max, 19200 baud. 
The main reason I am asking this is because the third device (on the software USART) has a provided network library, that I am frankly, not comfortable with modifying. 
The way the third device works is, from the point of calling code, is that you pass it a "/dev/tty*" serial device, and that is how the Pi communicates with the end device. 
The documentation somewhat implies that it is possible, but does not actually say how it works.
Is it possible to make pigpio treat a pair of GPIO's as the Rx/Tx for a /dev/* device?

joan2937, 2018-04-17:
I am afraid that is not possible.

You can use [gpioSerialReadOpen](http://abyz.me.uk/rpi/pigpio/cif.html#gpioSerialReadOpen) to read serial data from a GPIO.  To write serial data is more complicated and uses waves, in particular [gpioWaveAddSerial](http://abyz.me.uk/rpi/pigpio/cif.html#gpioWaveAddSerial).

The only simple option (presumably discarded) would be to use a USB serial dongle which I think would appear on `/dev/ttyUSB0`.

guymcswain, 2018-04-17:
**If** you are using nodejs, the `pigpio-client` package can create a 'pigpio bit-bang' serial port using arbitrary GPIO pins.  In your app you could create a named pipe, '/dev/*', that connects to the serial port object that pigpio-client created.

ShakenU, 2018-04-17:
@joan2937 Thanks for your reply. The reasons we are not going through that path is that A) The CM3 only has one USB, which is being used for something else, and the person designing the board couldn't get a USB hub to work properly. 
What I will end up doing, I think, is just creating a piece of middle-ware that presents itself as a /dev file and then any IO on the GPIO gets translated to the stream on the file.

closed 2018-04-17 by ShakenU