joan2937/pigpio #210: Pigpiod Automatic Failure

kkoppul2, 2018-04-18:
Hello there,

I am running Ubuntu Mate on a Raspberry Pi 3B and I need to use the pigpiod_if2 library to access the GPIO pins without sudo access. Unfortunately every time that I boot or restart the service, the daemon immediately stops. I am unable to find any relevant documentation about why it won't run, given the error messages provided. This is the systemctl status pigpiod.service output.

● pigpiod.service - Pigpio daemon
   Loaded: loaded (/lib/systemd/system/pigpiod.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/pigpiod.service.d
           └─public.conf
   Active: failed (Result: signal) since Tue 2018-04-17 22:00:12 CDT; 1s ago
  Process: 2004 ExecStop=/bin/systemctl kill -s SIGKILL pigpiod (code=killed, signal=KILL)
  Process: 1998 ExecStart=/usr/bin/pigpiod (code=exited, status=0/SUCCESS)
 Main PID: 1998 (code=exited, status=0/SUCCESS)

Apr 17 22:00:12 bilbot-desktop systemd[1]: Started Pigpio daemon.
Apr 17 22:00:12 bilbot-desktop systemd[1]: pigpiod.service: Control process exited, code=killed statu
Apr 17 22:00:12 bilbot-desktop systemd[1]: pigpiod.service: Unit entered failed state.
Apr 17 22:00:12 bilbot-desktop systemd[1]: pigpiod.service: Failed with result 'signal'.

I've set the public.conf file to match the one that is provided in the pigpiod_if2 repository, but it throws an error about multiple ExecStarts if I don't add an empty one in the file.

Any help would be appreciated.
Thanks,
Karun

guymcswain, 2020-01-08:
Not sure why this issue was closed because I can reproduce the problem using this repo's [pigpiod.service](https://github.com/joan2937/pigpio/blob/master/util/pigpiod.service) file currently containing:
```
[Unit]
Description=Pigpio daemon

[Service]
Type=simple
ExecStart=/usr/bin/pigpiod

[Install]
WantedBy=multi-user.target
```
I believe this is because Type=simple means the service will exit immediately after the fork is invoked.  The library begins to initialize but later received a "sigHandler: Unhandled signal 18, terminating" from systemd.  Signal 18 is SIGCONT which must be systemd's way to indicate the process can continue.

Until (or unless) pigpio's signal handler is changed to handle SIGCONT, `Type=forking` must be used in the service unit.  As follows:
```
[Unit]
Description=Pigpio daemon

[Service]
Type=forking
ExecStart=/usr/bin/pigpiod

[Install]
WantedBy=multi-user.target
```
Alternately, one could still use the 'simple' service type by using the '-g' option on pigpiod.  As follows:
```
[Unit]
Description=Pigpio daemon

[Service]
ExecStart=/usr/bin/pigpiod -g

[Install]
WantedBy=multi-user.target
```
Note that Type=simple is implied with ExecStart= and neither Type= nor BusName=.

closed 2020-01-08 by guymcswain

guymcswain, 2020-01-08:
Incidentally, if you run the service the 'wrong' way, change your service file and restart the service it will fail.  Apparently pigpio is left in a bad state from the unhandled signal.  

> Jan 08 18:23:53 pigpioV68 pigpiod[1656]:2020-01-08 18:23:53 initMboxBlock: init mbox zaps failed
Jan 08 18:23:53 pigpioV68 pigpiod[1656]: Can't initialise pigpio library

You need to reboot the raspberry pi.

commit d1fb6b92 (2020-01-08): Update pigpiod.service

Closes #210