joan2937/pigpio #212: Observed one core full load when writing to custom fifo through script

MaxDZ8, 2018-05-02:
I'm not sure of what's going on. I have been wrangling FIFOs a while today but this one leaves me ?_?. I'd have to walk through pigpio to understand what's going on but for the time being I'd just like to know if you can observe this as well.

# Reproduction

HW: Rpi 2B+
OS: Raspbian Stretch
Pigpio version: I forgot to pull this.

1. Have a console, let it be called 'conout'
2. working dir is /home/pi
3. `mkfifo target_pipe`
4. `sudo echo /home/pi/target_pipe w >/opt/pigpio/access`
5.  `sudo pigpiod`
6. `cat target_pipe`
7. Leave this alone.
8. Have a console, let it be 'concommand'
9. `pigs fo /home/pi/target_pipe 2` (result is '0' for me)
10. `pigs fw 0 115 116 97 114 116 58`
11. conout shows "`start:`", as expected
12. `pigs fw 0 115 116 97 114 116 58` (result is '0' for me)
13. `pigs procr 0`

Expected: conout should be updated with another `start:`.
Observed: nothing appears in conout; CPU load 25% not observed to terminate. Concommand is ready for other commands right away.

I looked at [fileWrite](https://github.com/joan2937/pigpio/blob/master/pigpio.c#L12968) and I cannot figure out why this behaviour should ever happen. Parsing isn't quite crystal-clear to me. Given my understanding of file handles in pigpiod (looks like they're all shared) I am without a clue on why this happens. Is this even supposed to work?

joan2937, 2018-05-02:
I can't see that working from a script.  If I remember correctly commands with more than three parameters will not work within a script.

guymcswain, 2020-01-02:
Could not reproduce this issue.  Observed `start:start:` in conout.  Never observed high cpu utilization.

Note.  4. `sudo sh 'echo /home/pi/target_pipe w >/opt/pigpio/access'`

closed 2020-01-02 by guymcswain