joan2937/pigpio #213: gpioHardwarePWM() pins and channel ?

Pluscrafter, 2018-05-05:
Is that right, that you only can use the pins 12,13,18,19 for HardwarePWM and are the pins 12,18/13,19 connected together ?

Pluscrafter, 2018-05-05:
I've tried that, but it has too much jitter and I've found, that I can use a pwm driver like that PCA9685, but this has too many outputs for my project and is there a pwm driver with only 4-6 outputs with 12bit resolution?

guymcswain, 2018-05-07:
The jitter issue on pigpio generated pwm has come up before.  Has anyone tried registering the pwm outputs to an external D-type flip flop using the pwm peripheral clock on the half phase?  Octal DFF devices are pretty cheap.

joan2937, 2018-05-07:
What are you driving?

I have only noticed jitter when I sample at 1MHz.  That hammers  the DMA bus with traffic.  I have not noticed jitter at lower sampling rates.

Of course jitter will be present, I'm just wondering if you can quantify the degree.

Pluscrafter, 2018-05-07:
I've connected one esc to the raspi and set gpioCfgClock(1,0,0); then I've set the samplerate to 250Hz and the resolution to 4000, witch is the max realResolution. Then I've control bldc motor with a pulse between 1000us and 2000us. when the throttle is at the lowest postiton the motor is shaking a little bit, but with hardwarepwm the the motor stops with 0% throttle and the motorspeed was stable.

guymcswain, 2018-05-10:
Have you tried gpioCfgClock(5,0,0)?

Pluscrafter, 2018-05-11:
yes but it also shake. But when I increment the values a little bit the servo is stable, but not 100% in the 0 degree position.

joan2937, 2018-05-13:
Is this a continuous rotation servo?  If so that sounds like a dead band calibration problem.

Pluscrafter, 2018-05-13:
no. Thats also the problem with my esc in the 0 Position it shakes a little bit, but with a little bit of throttle is has a stable rotation.

manek0, 2019-01-30:
How can I use PWM for pins > 31?
I have Compute Module 3 and would like to run PWM on pins 35-36. I can control these pins as usual GPIO, but I need PWM.

closed 2020-01-02 by guymcswain