joan2937/pigpio #216: pigpiod daemon make reboot or shutdown very long Raspberry 3 B+

xamix, 2018-05-17:
Hello everyone,

I'm on the last Raspberry 3B+ and last Raspbian Desktop.
_pigpiod_ is working for me on Python 3.

I tested with 

However when the dameon **pigpiod** is launched, then the reboot or shutdown or even power off take about 3minute instead of few seconds without demons. 

I'm connected with SSH and type **sudo poweroff** or **sudo reboot** and then it seem starting powering off, screen is black and in bottom left border it print **plymouth-reboot.service** for about 3 mintutes...

Do you have any idea about where come the problem?

Regards

guymcswain, 2018-05-19:
I run _pigpiod_ on zero-w, 1B, 1B+, 2, 3 connected via SSH and don't have this problem.  I'm using Raspbian Lite exclusively.

xamix, 2018-05-21:
Since it was only for control hardware PWM, I changed my program by using overlay and **sysfs**.
So I not use pigpiod anymore.

closed 2018-05-21 by joan2937