joan2937/pigpio #217: Example for creating a IR-signal

Joey-1970, 2018-05-19:
Hello!

May someone give me a example how to send a IR-signal on GPIO x with a IR-transmitter?
My idea is use the code for the IR-code that I find on the lirc-remotes database like this:

>  KEY_POWER                0x000000000000BCAF        #  Was: power
          input                    0x000000000000B6A5
          aspect                   0x0000000000007B68
          KEY_MUTE                 0x0000000000009E8D        #  Was: mute

I think I must at first create a "basic-wave" with 36kHz? How cound merge it with the code for a special command?

Joachim

guymcswain, 2018-05-19:
Hi Joachim,

You just need to generate a data waveform as described by the lirc lib.  I.E. send an array of pulse tripletts as described by the `waveAddGeneric()` API.  Follow the example in the doc:.  waveNew(), waveAddGeneric(), waveCreate(), then choose one of the waveTransmit() methods.

You connect the GPIO output to an IR LED that generates the carrier frequency (38Khz as I recall).  The data waveform modulates the carrier.

Also, check Joan's website as there may already be examples of this.  If not IR, 433 ook transmission is the same principle.
Cheers

Joey-1970, 2018-05-19:
Hello Guy!

I try to write down now what you write in "pigs":
1. WVCLR
2. WVAG
3. WVCRE
4. WVTX or WVTXR

The special IR code f.e.  0x000000000000BCAF I must put in the "trips" of WVAG?
When I look at this example (http://abyz.me.uk/rpi/pigpio/cif.html#gpioWaveAddGeneric) I must split the hex-number from above into a binary-number and add this to a array first andafter that as a triple to the "trips"?

I see differt frequencys, I don't konw what is right at this moment...

Joachim

guymcswain, 2018-05-19:
You've got the general outline correct.  The difficult part is converting the '1' and '0' bits into the required pulse format.  Then there is usually a preamble/postamble involved.  

Instead of using a library like lirc I just decided to create a waveform for a '1', save the returned wave id (wid) and do the same for a '0' and an 'f' if required.  Then to send a binary code just map the bits to the corresponding wid then send the array of wids.  I think the API used is called waveChainTx().

joan2937, 2018-05-21:
Here is some code I was working on ages ago.  I stopped when I realised there were thousands of IR protocols but it does cover RC5.

It is "as is".  Don't ask me any questions about it as I will not answer.

```python
#!/usr/bin/env python

import time
import pigpio

def bit_get(word, n):
   """
   Get bit n of word.
   The least significant bit is bit 0.
   """
   return (word>>n)&1

def bit_set(word, n, value):
   """
   Set bit n of word to value.
   The least significant bit is bit 0.
   """
   value = (value&1L)<<n
   mask = (1L)<<n
   return (word & ~mask) | value

def bits_get(word, lsb, msb):
   """
   Get bits lsb to msb of word.
   The least significant bit is bit 0.
   """
   mask = 2L**(msb + 1 - lsb) - 1
   return (word >> lsb) & mask

def bits_set(word, lsb, msb, value):
   """
   Set bits lsb to msb of word to value.
   The least significant bit is bit 0.
   """
   mask = 2L**(msb + 1 - lsb) - 1
   value = (value & mask) << lsb
   mask = mask << lsb
   return (word & ~mask) | value

def bits_reverse(word, lsb, msb):
   """
   Get bits lsb to msb of word and reverse them.
   The least significant bit is bit 0.
   """
   result = 0
   for i in range(lsb, msb+1):
      if (word >> i) & 1: result |= 1 << (msb - i)
   return result

def _carrier(gpio, frequency, micros, dutycycle=0.5):
   """
   Generate cycles of carrier on gpio with frequency and dutycycle.
   """
   wf = []
   cycle = 1000000.0 / frequency
   cycles = int(round(micros/cycle))
   on = int(round(cycle * dutycycle))
   sofar = 0
   for c in range(cycles):
      target = int(round((c+1)*cycle))
      sofar += on
      off = target - sofar
      sofar += off
      wf.append(pigpio.pulse(1<<gpio, 0, on))
      wf.append(pigpio.pulse(0, 1<<gpio, off))
[... 864 lines cut ...]

   #r = SamsungRead(pi, GPIO_RX)

   #time.sleep(60)

   GPIO_mode = pi.get_mode(GPIO)

   s1 = ir_protocol.RC6_0(pi, GPIO)
   s2 = ir_protocol.RC5(pi, GPIO)

   for a in range(256):
      s1.set_address(a)
      s2.set_address(a)
      for c in range(256):
         s1.send(c, 2)
         s2.send(c, 2)
         time.sleep(0.2)
         print(a, c)

   s.cancel()

   pi.set_mode(GPIO, GPIO_mode)

   pi.stop()
```

SlySven, 2018-05-21:
ℹ️ A former workmate of mine back when I was designing TV hardware for a living, created a bit of low-level PC (possibly assembler-code based) bit-bang software used in conjunction with a parallel port pin driving a simple transistor and IR LED hardware unit to do something like this. It was to enable us to simulate RC5 codes for the TVs we were creating on so that we could test the software he was writing for codes, that were not in the range that the existing remote-controls we had, possessed.

One feature you might want to ensure you have (that we found the hard-way) is that you ought to consider a.c. coupling (via a suitable, simple resistor-capacitor arrangement) the signal to the switching element that controls the current through the IR LED - so that a steady level does not turn the LED on.  They are normally driven from a source comprising a pair of AA/SUM3 cells though with an electrolytic capacitor to provide a bit more of a current pulse for a bright IR pulse with a waveform that has a low duty cycle.  If you are not careful with your software you can easily have the diode conducting for more than a fraction of a second if there are glitches or crashes - and if the unit is being supplied from anything like a decent power supply the normal LED series resistor can usually allow a significant part of an ampere of current through that emitter - which will not emit for *very* long under those conditions...

closed 2019-07-11 by joan2937