joan2937/pigpio #218: GPIO write at REAL_CLOCK

nachoplus, 2018-05-28:
Hello,

I am trying to generate a signal on GPIO pin at precise wall clock but callback+write() latencies are about 1300us. I discover trying to mirror a PPS signal (prove with python and C). 

I understand that latencies are unavoidables but I am wondering if I there are any solution of do precise gpio activation at time base on system clock (i.e each UTC second)

Thanks in advance

joan2937, 2018-05-29:
I have written software using pigpio try to do a similar job.  You could try searching for joan and GPS on the raspeberrypi.org/forums where I did post a couple of examples (years apart).  They might give you an idea, although I'm not sure if they would be compatible with the current pigpio (they depended on internal functioning).

I might have another look.

My approach would to be to use a wave.

Every second I'd send a wave looking like.
```
                         +-+
                         | |
                         | |
                         | |
    preamble             | |      postamble
-------------------------+ +---------------------------
```
The overall length would be a million (or whatever number of ticks calibrates to one second).
The low preamble and postamble could be altered to centre the pulse on the second.

Every second you would need to calibrate the system tick against wall time so that the pre-amble and post-amble could be changed if needed as well as the overall length.  The new wave could then be sent using the SYNC option (see http://abyz.me.uk/rpi/pigpio/cif.html#gpioWaveTxSend).

nachoplus, 2018-05-29:
Hi Joan,

I am not sure I understand you ... I mean gpioWaveTxSend()  will have some latency as write() has so... How do you asure that preamble pulse will be started at rigth time (or tick)?

One solution would be to make a new function to write gpio in the future at desire tick write_at_tick(gpio,level,future_tick) but I don't know if is fesasible..

Thank you for your anwers and your incredible support!

nachoplus, 2018-05-29:
OK. I understand your point. Using this aproach you can get a equal space in CLOCK_REALTIME  train of pulses.

But how do you know either the tick or the wall clock when the first wave start at? You need this data to do the first 'prologue' calculation. Don't you?

nachoplus, 2018-06-25:
I answer myshelf. Put in simply I used a  RISSING_EDGE callback on auxiliary signal to do wave to UTC syncronization. Now I am able to fire a gpio with an accuracy of +-5micros from UTC !!

closed 2018-06-25 by nachoplus