joan2937/pigpio #219: Using Pipe instead of Socket and RS485

Odianosen25, 2018-06-10:
Hello @joan2937,

Thanks for your work.

Please two questions I have:
1) How do I use the pipe over `/dev/pigerr`, if using it on the local machine instead of the socket on port `8888`? I want to do this, for security reasons where I plan on using your code. As I haven't seen any example unfortunately on how to do this.

2) Is there anyway for me to access like RS485 or something? I have this device on my Pi via the address `/dev/ttyRS485`, and since I want to use your code to access my other IOs on my Pi, thought will be nice I could do this also. if I can't please just confirm, so I seek another option; as I wouldn't want to be going down the rabbit hole, only to find out at the end its not possible.

Thanks and regards

joan2937, 2018-06-11:
To use the pipe interface you write commands to /dev/pigpio and read the response from /dev/pigout.

The commands are documented in [pigs](http://abyz.me.uk/rpi/pigpio/pigs.html) which has a few pipe examples at the start.

Also see x_pipe in the archive.

If you use the interface from C note that a line is processed when the newline is seen.

I know little of RS485.  You could try using the [serOpen](http://abyz.me.uk/rpi/pigpio/cif.html#serOpen) suite but It has no support for the control signals.

Odianosen25, 2018-06-14:
Hello @joan2937,

Many thanks for the response. I am trying to create an event for Serial, that is when I receive a data on either of my serial devices. 

There seems to only be the ability to specify callbacks for the GPIO, but I want to be able to declare for Serial also, then I can check what was sent, and read them. 

In case I am missing something, can you point me in the right direction how to do this? Basically declare a callback for a serial data available event.

I came across the `pi.event_callback(<event_id>, cbf)` where `event_id` within 1-31. I am assuming these event's only have to do with the GPIO as I can't find where its explained what these ids mean.

So I need one for serial please or I2C or anything outside just the GPIO.

Please forgive me if I am missing the obvious.

Kind regards

guymcswain, 2018-08-24:
It's a rabbit hole.  I've gone down it.

You can implement a mechanism to notify yourself when data is arriving on a gpio, wait for a short while, then call your process to go read the serial data from the buffer.  That's not the rabbit hole.  RS485 is implemented with protocols that determine when you switch from receiving to transmitting.  Usually, you have a fixed amount of time to turn the bus around.  Because you are running on a preemptive operating system you can't guarantee you'll meet the turn-around time requirement.

guymcswain, 2020-01-02:
Changed title from `Using Pipe instead of Socket and RS485` to `How to declare a callback for a serial data available event?`

guymcswain, 2020-01-25:
> In case I am missing something, can you point me in the right direction how to do this? Basically declare a callback for a serial data available event.

No explicit method is provided to do this and it is left to the user to handle at the application level.  I may one day implement a script (using the pigpio script API) to provide this capability and share it with the community.  Likewise, if others have an implementation they wish to share, I will include it on the project's wiki pages.

closed 2020-01-25 by guymcswain