joan2937/pigpio #220: Calling `exit` from a signal handler isn't safe

jonas-schievink, 2018-06-19:
Here's a snippet from inside `sigHandler`, which calls `exit` at two occasions:

https://github.com/joan2937/pigpio/blob/934874be2fa34a525beb33e8cb75e378df587860/pigpio.c#L5518-L5532

According to `man 7 signal`, this is not a safe thing to do. Only `_Exit` and `_exit` are async-signal-safe, not the regular `exit`. In my case, this causes my `atexit` handler to run in signal context, which is very unexpected.

(note that this might not be the only signal safety issue)

joan2937, 2018-06-19:
If exit is called in that function something pretty dreadful has already happened.  Continuing as if nothing is wrong may not be sensible, aborting is probably the right solution.

What would you do?  Other threads are already potentially compromised.

jonas-schievink, 2018-06-20:
The code is executed when the program is killed via `SIGINT` or `SIGTERM`, nothing is compromised in that case. You also shouldn't exit with a code of -1 in that case, which indicates an error.

I think it's best to just shutdown pigpio normally and *don't* call any exit function. The default action of the signal will then be executed and usually terminate the process.

guymcswain, 2020-01-02:
@jonas-schievink , Do you have a proposal for a change?  If so, we will evaluate it.

jonas-schievink, 2020-01-02:
Call _exit or _Exit instead of exit. I'm not qualified to review the handler for other signal safety issues though.

guymcswain, 2020-05-02:
The signal handling was implemented in support of the daemon (pigpiod).  You can turn off the signal handler if you don't want to use it:
```c++
gpioCfgSetInternals(PI_CFG_NOSIGHANDLER);
```

jonas-schievink, 2020-05-02:
The signal handler invokes undefined behavior. Please reopen until that is fixed.

guymcswain, 2020-05-02:
Are you using the pigpiod?

jonas-schievink, 2020-05-02:
I don't remember. It's still undefined behavior, whether it's in the daemon or not.

guymcswain, 2020-05-02:
Yes, but as Joan stated, if things are in the midst of crashing, does it matter?  I suppose I could make the change.  Have you tested it?

jonas-schievink, 2020-05-02:
> if things are in the midst of crashing, does it matter?

Unfortunately yes, or I wouldn't have opened this issue – having `atexit` handlers called from a signal handler is very unexpected behavior. In my case I think it was some ROS handler running and causing havoc.

> Have you tested it?

No, as I no longer have access to the hardware.

guymcswain, 2020-05-02:
Will you help me to write a test case?  I'd like some confirmation that whatever change is made to the source if having the intended behavior (and that nothing else breaks).

jonas-schievink, 2020-05-02:
I can try, but I'm not a C programmer. Basically you need to:

* Register an `atexit` handler
* Setup pigpio
* Send a SIGINT signal to yourself using `raise(SIGINT);`
  * This will now call the registered handler, but should not (technically it's UB, so the handler doesn't *have* to be called)

guymcswain, 2020-05-02:
Can you please clarify?

> , but should not (technically it's UB, so the handler doesn't have to be called)

jonas-schievink, 2020-05-02:
Yeah, `atexit` handlers are only called by `exit`, not the async-signal-safe alternatives `_exit` and `_Exit`.

joan2937, 2020-05-05:
I would not make any change to the existing behaviour.  Undefined behaviour is just that, an unknown number of people will be depending on the current behaviour.

closed 2020-06-28 by guymcswain