joan2937/pigpio #222: Waveform generation fails for pulse lists with more than 5460 elements

matthew-viglione, 2018-06-29:
When I call `wave_add_generic(pulse_list)` with more than 5460 elements in the pulse list, the function either hangs silently or displays `ConnectionResetError: [Errno 104] Connection reset by peer`

I cannot use wave_send_repeat() because in our application the high and low times will vary (We are generating motion curves to drive a high-precision micro-stepped motor). 

I am using Python 3.5.3. You can replicate the error with this code: 

```python
import pigpio
import time

OUPUT_PIN = 23      

pi = pigpio.pi()                        # initialize connection 
pi.set_mode(OUPUT_PIN, pigpio.OUTPUT)   # set pin direction
pi.wave_clear()                         # clear the waveform buffer             

periods = input("How many periods? ")   # prompt for number of periods 

print("Creating pulse list...")
pulse_list=[]                               # create an empty pulse list 

for i in range(0,int(periods)):             # for the specified number of periods 
    # turn OUPUT_PIN on for 10us
    pulse_list.append(pigpio.pulse(1<<OUPUT_PIN, 0, 10))     
    # turn OUPUT_PIN off for 10us
    pulse_list.append(pigpio.pulse(0, 1<<OUPUT_PIN, 10))      

print("Pulse list done. Pulses:", len(pulse_list))

print("Adding pulses to waveform...")
pi.wave_add_generic(pulse_list)             # add the list of pulses to a waveform

print("Creating waveform...")
pulse_waveform_id = pi.wave_create()        # create and save an id for the waveform 

print("Sending waveform...")
pi.wave_send_once(pulse_waveform_id)        # send the waveform 

print("Done.")
```
This will give the output: 

```
How many periods? 2731
Creating pulse list...
Pulse list done. Pulses: 5462
Adding pulses to waveform...
Traceback (most recent call last):
  File "example.py", line 27, in <module>
    pi.wave_add_generic(pulse_list)             # add the list of pulses to a waveform
  File "/usr/lib/python3/dist-packages/pigpio.py", line 2120, in wave_add_generic
    self.sl, _PI_CMD_WVAG, 0, 0, len(pulses)*12, extents))
  File "/usr/lib/python3/dist-packages/pigpio.py", line 1007, in _pigpio_command_ext
    dummy, res = struct.unpack('12sI', sl.s.recv(_SOCK_CMD_LEN))
ConnectionResetError: [Errno 104] Connection reset by peer
```

joan2937, 2018-06-30:
I think the maximum socket message is 65k.  You could change the constant in pigpio.h (I think) and rebuild.

A better solution would be to add less elements.  Say add 4000 at a time, starting each new block where the old one ended (ny making the first element a delay).

matthew-viglione, 2018-07-01:
How do I add the delay to get them to line up? What would that look like?

joan2937, 2019-08-01:
Commit the waveform so far before the message gets too big.

Have a look at the following example code.

```python
if len(pulse_list) > 4000:
   # commit waveform so far
   pi.wave_add_generic(pulse_list)
   pulse_list = []
   # start new waveform at end of last
   pulse_list.append(pigpio.pulse(0, 0, pi.wave_get_micros()))
```

closed 2019-08-01 by joan2937

AstraLuma, 2021-07-24:
We just hit this. Would you consider guard rails or a more helpful error?

AstraLuma, 2021-07-24:
Oh, and for anyone else: The constant is `CMD_MAX_EXTENSION`:

https://github.com/joan2937/pigpio/blob/c33738a320a3e28824af7807edafda440952c05d/command.h#L41

AstraLuma, 2021-07-24:
And this is the critical code, at least when using network connections (which is default when using the python library)

https://github.com/joan2937/pigpio/blob/c33738a320a3e28824af7807edafda440952c05d/pigpio.c#L7053-L7083

KubaO, 2025-03-20:
I think `wave_add_generic` should do the chunking by itself. At least there'll be a guarantee it's done right. Right now, every user has to reimplement that bit of code. It's wasted effort really. Would you accept a PR making `wave_add_generic` not have this gotcha, at least in Python API?