joan2937/pigpio #226: No edge detection callbacks received first time after a reboot

cosinekitty, 2018-07-13:
I'm running a Python3 program called `lidet.py` to capture edge detection events for a home-made radio receiver that detects lightning strikes. Everything works fine **except** the first time I run the program after rebooting the Pi. If I kill the program and run it again, it always works fine after that. 

So I ended up creating a hack script where I run the program with the command line parameter "hack" that signals it to immediately exit, then run it again without "hack" on the command line. With or without the "hack" parameter, I initialize pigpio, install the callback, and then stop pigpio. The only difference is that without "hack", the program blocks until killed so the callbacks can be processed. 

Anyway, with my workaround, everything works great, but it this is very confusing, especially for anyone trying to get this to work for the first time.

More details:

- All my code is visible at https://github.com/cosinekitty/lightning

- I'm using Python 3.2.3 on a Raspberry Pi Model B+.

- Rasbpian info:
```
don@curry ~/github/lightning $ uname -a
Linux curry 4.1.19+ #858 Tue Mar 15 15:52:03 GMT 2016 armv6l GNU/Linux
```

- I use `crontab -e` from the root user to start the daemon: `@reboot  /usr/local/bin/pigpiod`

- From my normal user account `don` I use `crontab -e` to run the script `runlogger`. See link above to read the code. This script waits 20 seconds before starting `lidet.py`, in an attempt to make sure the `pigpiod` daemon is up and running, since both the daemon and the script are launched at boot time. (I also tried a 60 second delay but the problem still happened.)

- The hack I mentioned in the `runlogger` script is the following line:
```
python3 lidet.py hack
```
If I comment out that line, the bug happens every time I reboot. I have tried this at least a dozen times both ways and it is 100% consistent.

Thanks in advance for any help on this...

joan2937, 2018-07-15:
It works for me.  Just copied the code to a Pi and simulated the input in another terminal window with `pigs w 21 0 w 21 1`.

The daemon only reports level changes.  If it sees the level as 1 it will not issue a report until the level has changed to 0.

Could you reboot then read the level (`pigs r 21`) then run the script without the hack and read the level again (`pigs r 21`).  I suppose the second read should be 1 as you have set a pull-up.  Could you then toggle the GPIO (`pigs w 21 0 w 21 1`) and see if the strike is recorded.

cosinekitty, 2018-07-15:
Thanks for looking into this!  That helped me figure out what was going on. I did have a bug where my callback caused an exception if I received the HIGH edge before the LOW edge, which would only happen the first time I ran my program after a reboot.

However, the reason I didn't figure this out on my own immediately was I didn't realize an unhandled exception in a pigpio callback would not terminate the program. I didn't see the exception output because this program was running as a cron job, so there was no visible console output. Once I ran lidet.py manually I could see the exception stack trace, but surprisingly the program keeps running.

Once the exception happens, my callback is never called again, no matter how many times I pulse pin 21. I think pigpio should terminate the whole program and exit with a nonzero value if any unhandled exception occurs in a callback.  That would be much better than leaving the program stuck in a non-functioning state.

joan2937, 2018-07-15:
I suppose you could argue the case but I am not going to change the Python module to attempt to detect such an error.

The error occurs because of a logic error in user code which needs to be dealt with by user code.  There is no way pigpio can tell if the intention is to propagate such errors to an outer level of the program.

cosinekitty, 2018-07-15:
OK, fair enough. Thanks for publishing and maintaining pigpio. It is exactly what I needed.

closed 2018-07-15 by cosinekitty