joan2937/pigpio #227: _433.py not capturing all messages

dwaynez, 2018-07-19:
I have switched from using gpioSetGetSamplesFunc in a C program to using pigpiod and the _433.py program to process RF433Mhz communication from some Chinese security devices. When I was doing it in the C program, I was capturing every signal sent, however, using PIGPIOD I seem to drop a lot of sends from these devices. I changed the startup parameters for pigpiod to add "-s 1 -b 2000". Any other suggestions for improving the capture rate?

joan2937, 2018-07-20:
Python is slower than C for glue logic as Python is interpreted.  By glue logic I mean all the if/then/elses/loops etc. which are all interpreted.  Of course the heavy lifting in Python is handled by calling C functions which, by definition, will be as fast as C.

Increasing the sampling rate will be counterproductive.  It is only useful if the edges you want to detect are shorter than the default sampling rate of 5µs.  Legitimate edges in RF 433 comms are probably around 100µs which would be about 10kbps.  All increasing the sample rate will do is pick up more noise which would need to be discarded.

If anything I would decrease the sample rate to -s 10.  Increasing the buffer size may help.

guymcswain, 2018-08-24:
Running _433.py on remote host and nothing else on RPi with pigpiod.

dwaynez, 2018-08-24:
I resolved my problem. Firstly, when I added the LORA to the RPI, it dropped the 5V coming from the RPI and the RF433 was much less sensitive to signals. Before I discovered this problem, I had rewrote the receiver code into C. I really think it might have worked if I would have just addressed the 5V issue.

closed 2018-08-24 by dwaynez