joan2937/pigpio #232: pigpio.bb_serial_read returns wrong bytes at high baud rates

petronny, 2018-08-29:
In [this post](https://www.raspberrypi.org/forums/viewtopic.php?p=694626#p694626), qtfp said

>There is no error receiving from pigpio.wave_send_once to pc at baud:115200.
But some error receiving from pc to pigpio.bb_serial_read at baud:115200.
Also, I am wondering if pigpio will interfere with ttyAMA0.
It's really interesting and powerful!! Each port on pi can send and receive ttl!

And I have seen your reply:

> No, it will not interfere with ttyAMA0, unless you happen to be using gpio 14 or 15 for software serial.
I'd expect transmit to be more reliable than receive at high data rates. That's a function of the incoming bit sample rate (the PC end uses dedicated hardware).

Actually I build a similar test system today, using the Tx, Rx on Pi 3B (P14, P15) as the hardware serial,
P23 P24 as the software serial and almost the same python script.
I get similar results just like the post above and [here](https://raspberrypi.stackexchange.com/questions/27488/pigpio-library-example-for-bit-banging-a-uart).
pigpio has no problem when sending the serial waves, but will receive wrong bytes at high baud rates.

Is there any suggestion about fixing or debugging this problem?

petronny, 2018-08-29:
And I write a new script to send `b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'` every 3s to the software serial.
The outputs of `pigpio.bb_serial_read` are listed below.
PS. I have added the `ten_char_time` part to join bytes in a line just like your script.
Many of the lines are incorrect and they are almost untrustable.
```
b'\xbb\x01\x03\x00\x10\x00M100 20 Bm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M1\xb00 20dB- V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00\r100 20dB- V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dB- V2.0\xcd>'
b'\xbb\x01\x03\x00\x10\x00M100 20d\x02m V2.0\x8d>'
b'\xbb\x01\x03\x00\x10\x00M100 20dB- V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20 Bm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.\xb8\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V\xba.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dB- V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d>'
b'\xbb\x01\x03\x00\x10\x00\xcd\xff100 20dBm V2.0\x8d~'
b'\xfb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xfb\x01\x03\x00\x10\x00M100 20dC- V2.0\x8d~'
b'\xfb\x01\x03\x00\x10\x00M100 20dBm \x162.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dB- V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xff\x01\x03\x00\x10\x00M100 20 Bm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20$\x02m V2.0\x8d>'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\xcd~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V\x12.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20d\xc2\xff- V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20 Bm V2.0\x8d>'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm \x9e2.0\xcd~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d>'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm \x162.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xfb\x01\x03\x00\x10\x00M100 20dBm \x122.0\xcd~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20d\x02m V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20 Bm V2.0\x8d~'
b'\xbb\x01\x07\x00\x10\x00M100 v0dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dB- V2.0\x8d~'
b'\xbf\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xfb\x01\x03\x00\x10\x00M100 20dBm \x122.0\x8d~'
b'\xbb\x01\x03\x80\x10\x00M100 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xfb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\xcd~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 \xba0dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm \x122.0\x8d~'
b'\xfb\x01\x03\x00\x10\x00M100 20dBm V2.p\x8d~'
b'\xbb\x01\x03\x00\x10\x00\r100 20dBm \x122.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20d\x02m V2.0\xcd~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20d\x02- V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2\xae0\x8d~'
b'\xfb\x01\x03\x00\x10\x00M100 20d\x02m \x122.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M10\xb8 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00\r100 2\xb0dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm \x122.0\xcd~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm \x122.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00\r100 2\xb8`B\xed \x162.0\x8d~'
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~'
```

guymcswain, 2018-08-31:
Have a look at closed issue #166.

petronny, 2018-08-31:
@guymcswain Thanks for your reply.
I figured out that I need to add `-s1` to pigpiod to rise the sample rate.
I re-run it for 100 times, 94% of them are correct now.
All those incorrect lines are cutting the message incorrectly, like:
```
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~' True 50 0 50
b'\xbb' False 50 1 51
b'\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~' False 50 2 52
b'\xbb\x01\x03\x00\x10\x00M100 20dBm V2.0\x8d~' True 51 2 53
```
Then I add `-b1000` to pigpiod to make the buffer bigger.
98% of another 100 tests are correct now.
But there are still 2 of them cutting the message incorrectly.

So is there a way to avoid this?

PS. I've add `time.sleep(0.2)` to wait for more data.

guymcswain, 2018-09-01:
It is not possible to avoid these failures.  Issue #35 points to an inherit 'jitter' in the sampling rate.  You must accept that limitation and deal with it in your application - such as detecting error using checksums and retry transmission.

petronny, 2018-09-02:
@guymcswain Thanks again.
Fine... I will handle these errors manually.

closed 2018-09-02 by petronny