joan2937/pigpio #236: Windows socket interface

davidovv, 2018-09-10:
I tried to make modifications to pigpiod_if2 interface and had success with it working on windows, can you check it out and extend the library 
in pigpiod_if2.c i excluded/included following headers for windows version
```
#ifdef _WIN32
#include "winsock2.h"
#include "ws2tcpip.h"
#else
#include <netdb.h>
#include <sys/socket.h>
#include <netinet/tcp.h>
#include <sys/select.h>
#include <arpa/inet.h>
#endif
```

in function stop_thread pthread_join was giving me invalid parameter passed to c function so i excluded it, i am not sure about consequences...
```
#ifndef _WIN32
      pthread_join(*pth, NULL);
#endif
```
in pigpio_stop function changed close function to closesocket
```
#ifdef _WIN32
      closesocket(gPigCommand[pi]);
#else
      close(gPigCommand[pi]);
#endif
```
and
```
#ifdef _WIN32
       closesocket(gPigNotify[pi]);
#else
       close(gPigNotify[pi]);
#endif
```

guymcswain, 2018-11-14:
How important is it for you to have native bindings to the socket interface?  I use Windows machine to develop all my pigpio applications but use nodejs.  Python and other languages are supported.

guymcswain, 2020-01-02:
Aside from extending pigpiod_if2.c for Windows, are you indicating there is an issue with `close()` vs `closesocket()`?

Also, this is a big red flag:

> in function stop_thread pthread_join was giving me invalid parameter passed to c function so i excluded it, i am not sure about consequences...