joan2937/pigpio #241: Problems with bscxfer

matherp, 2018-09-29:
I'm trying to use the Pi as a i2c slave. I can send data to the Pi and also return data to the host but I'm struggling with the return data being one message behind - I'll explain...

```
void ProcessRemote(void){
    if(Option.Remote==0)return;
    int i=0, count,s1,s2;
    xfer.control = (0x13<<16) | 0x305;
    xfer.txCnt = 0;
    s1=bscXfer(&xfer);
    if((count=xfer.rxCnt)){
       for(i=0;i<count;i++){
            RemoteRxBuf[RemoteRxBufHead]=xfer.rxBuf[i];
            RemoteRxBufHead = (RemoteRxBufHead + 1) % remote_buf_size;     // advance the head of the queue
            if(RemoteRxBufHead == RemoteRxBufTail) {                           // if the buffer has overflowed
                 RemoteRxBufTail = (RemoteRxBufTail + 1) % remote_buf_size; // throw away the oldest char
            }
        }
        xfer.txBuf[0]=count & 0xFF;
        xfer.txBuf[1]=(count >>8) & 0xFF;
        xfer.txCnt=2;
        s2=bscXfer(&xfer);
        PIntHC(s1);PIntHC(s2);MMPrintString("\r\n"); //print values of s1 and s2 to the console in hex
    }
}
```

ProcessRemote is called frequently in the main program. When rxCnt is positive it processes the message into a circular buffer. It then returns to the host the number of bytes received. This all works except that if the host sends messages of length 12,30,12,30 it receives 0,12,30,12. i.e. the return message is always one message behind. xfer is defined globally.

The status returns from the two calls (s1 and s2) are 0x12 and 0x20042

I'm obviously missing something but what? All ideas gratefully appreciated.

alexhelms, 2019-09-11:
I am seeing a similar issue. I have a Pi 3 Model B using pigpio version V71 with python 3.7.3.

I'm just sending a couple bytes:
`pi.bsc_i2c(I2C_ADDR, bytearray([1,2,3,4,5,6,7,8,9,0xA])`

But what I am getting is offset by one:
`0A 01 02 03 04 05 06 07 08 09`

If I read more than 10 bytes, the rest past 0x9 are just 0x0A
`0A 01 02 03 04 05 06 07 08 09 0A 0A 0A 0A`

I would expect data past what was sent to be zeros but I don't see that. Also, the first byte seems to be related to whatever was the last byte in the buffer from the last send.

Anybody else seeing similar behavior?

guymcswain, 2020-06-28:
If anyone believes there is an issue with the library please open an issue and provide a simple test case to reproduce.

closed 2020-06-28 by guymcswain

matbs24, 2021-09-30:
I would use a Rpi 3B+ as I2C slave. I had the same issue explained here, what I want send back to master is always one byte behind of the requested. 

I did a quick test to explain this:
**I2C communication**
**Master**: Arduino uno sending incremental byte every second. Starting from value 1
```

#include <Wire.h>

void setup()
{
   Wire.begin(); // join i2c bus as master
   Serial.begin(9600);  // start serial for output
   while(Serial.available() == 0) {
    }
   int mydata = Serial.read();
}

char strOut[17],strIn[17];

int x = 0;

   
void loop()
{   
  while (x < 20){
     if (++x > 9999999) x=0;
     sprintf(strOut, "Sending %d \n", x);
     Serial.print(strOut);
     
     Wire.beginTransmission(24); // transmit to device #24 = 0x18
     Wire.write(x);           // sends 16 bytes
     Wire.endTransmission();    // stop transmitting
     
     Wire.requestFrom(24, 1);    // request 6 bytes from slave device #8
  
     while (Wire.available()) { // slave may send less than requested
        char c = Wire.read(); // receive a byte as character
        sprintf(strIn, "Received %d \n", c);
        Serial.print(strIn);
     }
     delay(1000);
  }
  
}
```

**Slave**:Rpi 3B+ running pigpio deamon at 1 us. Acting as I2C slave. When one byte read request arrived replay back with the same byte

```
import time
import pigpio

SDA=18
SCL=19

_slave_address = 0x18

def i2c(id, tick):
   global pi

   s, b, d = pi.bsc_i2c(_slave_address)
   global stop
   if b:
        data = int.from_bytes(d,"big")
        print(data)
        print(pi.bsc_i2c(_slave_address,[data]))
        stop = stop+1
        
if __name__ == '__main__':

    pi = pigpio.pi()

    stop = 0

    if not pi.connected:
        exit()

    # Add pull-ups in case external pull-ups haven't been added

    pi.set_pull_up_down(SDA, pigpio.PUD_UP)
    pi.set_pull_up_down(SCL, pigpio.PUD_UP)

    # Respond to BSC slave activity

    e = pi.event_callback(pigpio.EVENT_BSC, i2c)

    pi.bsc_i2c(_slave_address) # Configure BSC as I2C slave

    time.sleep(25)

    e.cancel()

    pi.bsc_i2c(0) # Disable BSC peripheral

    pi.stop()

```
[image]