joan2937/pigpio #247: Packaging pigpio in Debian.

plugwash, 2018-11-08:
I have been looking at getting pigpio into Debian (as part of the broader goal of getting the raspberry pi education software stack into Debian). I will probablly start from the raspberry pi foundation's deb packaging and try to improve it to meet Debian standards.

Firstly there seem to be a number of components to pigpio, I understand they fit together as follows. Please tell me if my understanding is correct.

pigpio: library for accessing the gpio directly, used by pigpiod
pigpiod: daemon for accessing the GPIO
pigs: command line client for accessing the GPIO via the daemon
pigpiod_if: deprecated library for accessing the GPIO via the daemon
pigpiod_if2: library for talking for accessing the GPIO via the daemon
pig2vcd: tool for converting data for use with gtkwave, not clear to me exactly where the input data comes from.
pigpio.py: python module for accessing the GPIO via the daemon.

Is this understanding correct?

Secondly your libraries don't seem to have sonames, would you object to a patch adding them?

Finally I notice that your code doesn't seem to have any checks that it is actually running on a Pi, would you object to a patch adding such checks?

joan2937, 2018-11-08:
Your understanding is correct and I have no problem with your improvements.

I always thought pigpio a bit too niche for Debian.

plugwash, 2018-11-08:
> I always thought pigpio a bit too niche for Debian.

The main reason we want pigpio is to provide remote gpio support for gpiozero, allowing educational materials that use gpiozero to be run on a Debian PC with a Pi acting as an IO interface.

plugwash, 2019-02-02:
After doing a bunch of device detection and 64-bit cleanliness fixes ( https://github.com/joan2937/pigpio/pull/255 ) I came to the conclusion that the pigpio server is not compatible with Debian kernels because the userspace mailbox interface is not present. The mailbox interface seemed pretty fundamental to the operation of the pigpio server side.

As a result of this I decided to package only the client side of pigpio in Debian at this time. The package of the client side stuff was accepted and has now migrated to testing. 

Thanks for your help.

guymcswain, 2020-01-02:
@plugwash , What is the impact of this?
> ... the userspace mailbox interface is not present.

I believe we would want to fix an issue that was *fundamental to (its) operation*.

plugwash, 2020-01-10:
The impact is that pigpio in it's present form will not work with upstream kernels. If we want pigpio to work with upstream kernels, then either the upstream kernel developers need to be persuaded to add the userspace mailbox interface (which may be a hard sell, I get the impression that the attitude of upstream kernel devs is that userspace should not be accessing hardware directly) or pigpio needs to be re-engineered to provide basic functionality without relying on any mailbox communication with the GPU.

joan2937, 2020-01-10:
A couple of years back I did start work on a replacement trying to learn from all the pigpio "mistakes".  The trigger was the new /dev/gpiochip interface and the expectation that the new Pis would be incompatible with pigpio (i.e. not based on a Broadcom SoC).

I did get quite far and had a working system.  It would have lost the pigpio specifics like DMA event capturing and generation.  It would have added various levels of security (SSH/MD5 messaging, finer grained permission control over peripherals such as GPIO, I2C, scripting etc.)

Unfortunately life intervened and I put it aside, perhaps the time has come to have another look.

guymcswain, 2020-01-13:
> ... or pigpio needs to be re-engineered to provide basic functionality without relying on any mailbox communication with the GPU.

This is not going to happen.  Much of the benefits of pigpio over other gpio libraries is the ability to access the DMA controller and other peripherals directly to enable monitoring of gpio without polling.  

Thanks for trying, however.

closed 2020-01-13 by guymcswain