joan2937/pigpio #254: Unhandled signal 11, terminating error

ajjsss, 2018-11-28:
I have updated pigpio to version 67 and attempted to remove the signal handling to prevent the following error as suggested in PR#164.
 
I tried adding the below code before gpioInitialise and still getting sigHandler: Unhandled signal 11, terminating error.   Any suggestions on what I am doing wrong?
gpioCfgInternals(1<<10, 0);

SlySven, 2018-12-02:
FWIIW: Signal 11 is a SIGSEGV - i.e. a segment violation - typically a use of a null pointer or something like that which tries to access an invalid part of the memory...

Unfortunately there are some aspects of the signal handling that are not as good as they could be *IMVHO*, IIRC with the default signal handing code pigpio will not ever do a *core-dump* so it can be harder to do a post mortem to determine what the state of things were when things crashed - I had a bash at this in my unmerged PR #58 but that was a bit much to get in I guess...

ajjsss, 2018-12-17:
Thanks for your comments SlySven. After reviewing,  still unclear what direction I might take to resolve this issue.

rosariosanto, 2019-01-07:
I have a similar problem. I'm using mono to execute a C# application in linux (ubuntu). The application ran for 10 to 20 minutes without issue, and then it terminated by the handle signal 11 or 30. So far, no solution to the problem. I tried many different things. My next move will be to use python instead of C#. The C# is using a wrapper around the pigpio library.

guymcswain, 2019-01-07:
> Any suggestions on what I am doing wrong?
gpioCfgInternals(1<<10, 0);

Try `gpioCfgSetInternals(1<<10)`.  According to docs, `gpioCfgInternals()` is deprecated

petercwallis, 2019-04-13:
Yes after a lot of trial and tribulation, this is the solution to the Signal 11 issue:

before calling gpioInitialise(),  turn off the signal handing by calling
gpioCfgSetInternals(1<<10)

This sets bit 10 whereas bgpioCfgInternas(1<<10,0) unsets it, but Joan inverted the bit's meaning apparently.

lionresearch, 2019-06-18:
I am also dealing with a Signal 11 issue that pops up after running a loop in my code for some time.  My error sounds very similar to above posts; my code runs just fine for a while and then dies.

I am trying to understand why using `gpioCfgSetInternals(1<<10)` may fix the issue.  Is the Signal 11 not actually a problem?  Or is it coming from another program but pigpio picks it up?  If it is something in my code causing the Signal 11 and I choose to ignore it, I would assume sometime later I will experience some sort of issue related to a segment violation. 

Thanks for the help

petercwallis, 2019-06-18:
I don't understand where this signal is coming from, but I suspect it is a
signal that is handled somewhere in the execution stack, but the pigpio
throws it before it gets to the place where it ought to be handled.  For
instance, it is very common in java to do:
 try{ Thread.sleep(...); } catch(InterruptedException e){}
If pigpio is intercepting the Interruption before the catch (which does
nothing) then there is no problem ignoring the signal11.

I have been using this approach in a modified version of pigpioj for a
while now without issues.  Indeed I think Mat has incorporated the solution
in the latest version of pigpioj.
P

lionresearch, 2019-06-18:
I gave the gpioSetCfgSetInternals(1<<10) a shot but I still get the problem.  I am running a user supplied daemon, and gpioInitialise is actually called in two different child processes, rather than a single time in the daemon.  I placed the gpioSetCfgSetInternals(1<<10) above gpioInitialise in both child processes.  Should the gpio functions actually be called from the daemon?  In Joan’s implementation it looks like she calls them from the daemon, pigpiod.  I saw on one forum Joan posted that calling gpioInitialise multiple times should not be a problem.

guymcswain, 2019-06-18:
@lionresearch what you describe is confusing.  Perhaps you could better explain what you are trying to do and your implementation.  There are two ways to access the pigpio functionality: The first by linking the pigpio library in your program, the alternate way is to access pigpio through the daemon using a socket interface library.  If you use the latter, you would not be using gpioInitialise since the daemon has already done so.  Instead you would be using gpioStart and gpioStop from the [pigpiod C IF](http://abyz.me.uk/rpi/pigpio/pdif2.html).

FWIW, I'm approaching 10 machine years of running pigpio using a socket interface and have never encountered issues with unhandled signals.   This may be your preferred approach.

lionresearch, 2019-06-18:
@guymcswain sorry for the confusion, let me try again:
We are linking the pigpio library into our program, which we launch with a daemon.  As written the daemon itself does not call gpioInitialise.  Instead, gpioInitialise() is called in two separate children processes that are launched by the daemon.

However, I just changed around the code so that the daemon calls gpioSetCfgSetInternals(1<<10) followed by gpioInitialise.  And I get the exact same behavior.  

The odd thing is my code will loop exactly 100 times and then die.  The behavior is the same every single time.  I am watching memory usage and processes and they do not appear to be growing at all.

My exact code in my daemon is:
 ```
gpioCfgSetInternals(1<<10); 
  // initialize gpio
  if (gpioInitialise()<0) {
    printf("GPIO failed to initialize\n");
    return 1;
  } else {
    printf("GPIO initialized.\n");
  }///
```

guymcswain, 2019-06-18:
> my code will loop exactly 100 times and then die

Fortunately, your case if very reproducible.  Should be easy to debug.  Good luck.

lionresearch, 2019-06-18:
@guymcswain I hope so!  No luck thus far, but yes I am quite happy I managed to write a very simple script that kills it the same way every time.

lionresearch, 2019-06-19:
I have narrowed it down to this line of code where my program produces signal 11 after it calls a certain task 100 times.  In other words, the code is controlling test equipment that can take a couple different types of measurements and when I make one type of measurement 100 times the code crashes.  

`(*(self->valueHandler_p))(self, &temp_meas);`

It does this every time despite setting different delays or calling other measurement types between the one that kills the program.  This all seems to scream that maybe an array is filling up or something just needs to be purged.  This is not my code and I know very little about C, but I as understand it this line of code has to do with pointers and maybe an array.  I think I can get my head around that with enough googling, but this line of code to me looks like:  (value)(value) and I cannot really figure that out.

thanks for any help, or any additional guidance on if you think this is looking like it is related this specific piece of code and not pigpio in someway.

guymcswain, 2019-06-19:
@lionresearch , please don't take any of my criticism the wrong way but I see several red flags in reading your comments.  Specifically:

>  This is not my code ...
>  ... gpioInitialise() is called in two separate children processes that are launched by (a) daemon
>  … I know very little about C

I myself don't know C well enough to help you out of the situation you are in.  So I would strongly urge you to re-evaluate why you chose the path you are going down.  It looks like you are trying to control some test equipment.  Great.  Depending on the interface and protocol there are surely many libraries available written in a language that you feel is your core competency.  If you want to use pigpio there are several clients available - Python, Java, NodeJs and others.  Said another way, there is no good reason to use pigpio library in a C program unless you have found that you can't solve your use case (control a piece of equipment to make measurements) any other way.  

To more specifically answer your question, yes the line of code ```(*(self->valueHandler_p))(self, &temp_meas);``` can cause bad things to happen if the function pointer is out of range or null.

lionresearch, 2019-06-19:
@guymcswain --no worries.  I'm in over my head but that's how we learn, right?  Thanks for the help thus far.

lionresearch, 2019-06-19:
@guymcswain sure enough, in a header file a static variable was assigned to 100 and it basically blows up the program because there is nothing in the corresponding C file resetting a counter.  Hey, at least I'm just the messenger in this case and learned a little about C programming along the way.

Also pigpio keeps on rocking along so I do not believe I have a need for `gpioSetCfgSetInternals(1<<10)` but I will keep it in mind if I see problems in long term testing of the measurement device.

guymcswain, 2020-01-03:
Duplicate of #138 which was fixed in V65.

closed 2020-01-03 by guymcswain