joan2937/pigpio #256: pigpio daemon inactive, even though connection still exists

jogientertainments, 2019-01-06:
Hi,
i'm using your library with a raspberry pi 3 model B+ and python. On the pi is running uWSGI and nginx. I'm also using flask.

I've got a problem with the pigpio daemon. If i start the daemon after boot, `systemctl status pigpio` outputs the following:
`pigpiod.service - Daemon required to control GPIO pins via pigpio
   Loaded: loaded (/lib/systemd/system/pigpiod.service; disabled; vendor preset: enabled)
   Active: inactive (dead)`
I'm pretty new to all this, but it seems to be not ok.
After that i import pipgio in python and connect to the daemon (i was just trying. Never expected, that it would work). The command `pigpio.pi().connected` outputs _True_. That seems to be ok, quite apart from the fact that the daemon is not active. If i type `pigpio.pi().stop()`, the connection is still active. This behaviour i don't understand.
My question is, if i missunderstand the usage of the library.
In addition, the real reason for this post is, that i want to control GPIO's. For that i've written the following code. 
```
import pigpio
from flask import Flask
from flask import render_template
from time import sleep
import RPi.GPIO as GPIO
import sys
from smbus2 import SMBus
import serial

@app.route("/forward/")
def forward():
    try:
        status = 0
        endstatus = 1

        while endstatus == 1:
            response = "3"

            if response == "2":
                status = int(response)
            elif response == "1":
                status = int(response)
            elif response == "3":
                status = int(response)

            # print(status)

            if status == 3:
                GPIO.output(Input3A, True)
                GPIO.output(Input4A, True)
                servo = 3
                pi = pigpio.pi()
                pi.set_mode(servo, pigpio.OUTPUT)
                pi.set_servo_pulsewidth(servo, 1300)

            if status == 1:
                GPIO.output(Input3A, False)
                GPIO.output(Input4A, False)
                pi = pigpio.pi()
                servo = 3
                pi.set_servo_pulsewidth(servo, 1500)
                pi.stop()
                endstatus = 0

    except KeyboardInterrupt:
        pi = pigpio.pi()
        all_off()
        pi.stop()

if __name__ == "__main__":
    app.run(host="0.0.0.0", port=8080)
```

With that code i can run it only once. If i click on the the corresponding button to execute the try block again, nothing happens. After that the uWSGI-log outputs:
```
(venv) root@raspberrypi:/home/pi/arc# tail /var/log/uwsgi/test_app_uwsgi.log 
  File "/home/pi/arc/venv/local/lib/python2.7/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/arc/movement.py", line 90, in forward
    pi.set_mode(servo, pigpio.OUTPUT)
  File "/home/pi/arc/venv/local/lib/python2.7/site-packages/pigpio.py", line 1340, in set_mode
    return _u2i(_pigpio_command(self.sl, _PI_CMD_MODES, gpio, mode))
  File "/home/pi/arc/venv/local/lib/python2.7/site-packages/pigpio.py", line 989, in _pigpio_command
    sl.s.send(struct.pack('IIII', cmd, p1, p2, 0))
AttributeError: 'NoneType' object has no attribute 'send'
[pid: 321|app: 0|req: 2/2] 127.0.0.1 () {50 vars in 901 bytes} [Sun Jan  6 14:39:40 2019] GET /forward/ => generated 0 bytes in 170 msecs (HTTP/1.1 500) 0 headers in 0 bytes (0 switches on core 0)
```
and
```
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio at localhost(8888)

Did you start the pigpio daemon? E.g. sudo pigpiod

Did you specify the correct Pi host/port in the environment
variables PIGPIO_ADDR/PIGPIO_PORT?
E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888

Did you specify the correct Pi host/port in the
pigpio.pi() function? E.g. pigpio.pi('soft', 8888)
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
```

I think it could be a port problem. But i didn't get the pigpio daemon configured to listen on i.e. port 8889. Starting with `pigpiod -p 8889 start` seems not to be enough.

If i try to start the daemon with the following command `sudo pigpio -l -n 127.0.0.1` the uWSGI log outputs:
```
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio at localhost(8888)

Did you start the pigpio daemon? E.g. sudo pigpiod

Did you specify the correct Pi host/port in the environment
variables PIGPIO_ADDR/PIGPIO_PORT?
E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888

Did you specify the correct Pi host/port in the
pigpio.pi() function? E.g. pigpio.pi('soft', 8888)
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
[pid: 320|app: 0|req: 2/2] 127.0.0.1 () {52 vars in 994 bytes} [Sun Jan  6 14:53:49 2019] GET / => generated 770 bytes in 5 msecs (HTTP/1.1 200) 2 headers in 80 bytes (1 switches on core 0)
[pid: 320|app: 0|req: 3/3] 127.0.0.1 () {48 vars in 846 bytes} [Sun Jan  6 14:59:50 2019] GET / => generated 766 bytes in 8 msecs (HTTP/1.1 200) 2 headers in 80 bytes (1 switches on core 0)
[pid: 320|app: 0|req: 4/4] 127.0.0.1 () {52 vars in 994 bytes} [Sun Jan  6 14:59:50 2019] GET / => generated 766 bytes in 8 msecs (HTTP/1.1 200) 2 headers in 80 bytes (1 switches on core 0)
/home/pi/arc/movement.py:54: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(Input1A, GPIO.OUT)
/home/pi/arc/movement.py:58: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(Input1B, GPIO.OUT)
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio at localhost(8888)

Can't create callback thread.
Perhaps too many simultaneous pigpio connections.
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Traceback (most recent call last):
  File "/home/pi/arc/venv/local/lib/python2.7/site-packages/flask/app.py", line 2309, in __call__
    return self.wsgi_app(environ, start_response)
  File "/home/pi/arc/venv/local/lib/python2.7/site-packages/flask/app.py", line 2295, in wsgi_app
    response = self.handle_exception(e)
  File "/home/pi/arc/venv/local/lib/python2.7/site-packages/flask/app.py", line 1741, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/arc/venv/local/lib/python2.7/site-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/arc/venv/local/lib/python2.7/site-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/arc/venv/local/lib/python2.7/site-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/arc/venv/local/lib/python2.7/site-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/arc/venv/local/lib/python2.7/site-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/arc/movement.py", line 90, in forward
    pi.set_mode(servo, pigpio.OUTPUT)
  File "/home/pi/arc/venv/local/lib/python2.7/site-packages/pigpio.py", line 1340, in set_mode
    return _u2i(_pigpio_command(self.sl, _PI_CMD_MODES, gpio, mode))
  File "/home/pi/arc/venv/local/lib/python2.7/site-packages/pigpio.py", line 989, in _pigpio_command
    sl.s.send(struct.pack('IIII', cmd, p1, p2, 0))
AttributeError: 'NoneType' object has no attribute 'send'
[pid: 320|app: 0|req: 5/5] 127.0.0.1 () {50 vars in 901 bytes} [Sun Jan  6 15:02:38 2019] GET /forward/ => generated 0 bytes in 165 msecs (HTTP/1.1 500) 0 headers in 0 bytes (0 switches on core 0)
```

I did a lot of research, but i'm stuck.

guymcswain, 2019-01-07:
Why `import RPi.GPIO as GPIO` ?

guymcswain, 2019-01-07:
What does `netstat -tulpn` report?

guymcswain, 2019-01-07:
Your looking for 127.0.0.1:8888

jogientertainments, 2019-01-07:
netstat -tulpn reports the following
```
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 127.0.0.1:2947          0.0.0.0:*               LISTEN      1/init              
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      541/nginx: master p 
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      530/sshd            
tcp6       0      0 ::1:2947                :::*                    LISTEN      1/init              
tcp6       0      0 :::22                   :::*                    LISTEN      530/sshd            
tcp6       0      0 :::8888                 :::*                    LISTEN      818/pigpiod         
udp        0      0 0.0.0.0:68              0.0.0.0:*                           497/dhcpcd          
udp        0      0 0.0.0.0:50270           0.0.0.0:*                           320/avahi-daemon: r 
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           320/avahi-daemon: r 
udp6       0      0 :::5353                 :::*                                320/avahi-daemon: r 
udp6       0      0 :::51032                :::*                                320/avahi-daemon: r
```

I already read, that there's a problem with tcp6. But i also don't know how to change it.

guymcswain, 2019-01-07:
My setup reports the same: `tcp6       0      0 :::8888` and I'm connecting with ipv4 address so this shouldn't be your problem.

> Perhaps too many simultaneous pigpio connections.

How are you sure this is not the problem?  I'm unfamiliar with flash, uwsgi (and Python is not my core language).

guymcswain, 2019-01-07:
Some suggestions:

- place `pi = pigpio.pi()` outside the 'forward' function definition.
- or, place `pi.stop()` at the end of each `if status == x:` block.