joan2937/pigpio #257: How to dynamically change spi mode and channel

davthomaspilot, 2019-01-12:
Not really an issue, but a "howto".  Not sure where to ask these kinds of questions.  Sorry if they don't belong here, but a suggestion on where would be welcome.

I have two SPI slaves, one on CS0 and the other on CS1.  

One slave requires SPI mode 2, while the other can be configured as a slave in either mode 0 or mode 1.

So, I have to dynamically change the spi mode when switching slaves.

Do I do this by doing an spiClose, then SpiOpen?  What happens to the SCLK, and MOSI, and the CS lines after the spiClose but before the spiOpen?

davthomaspilot, 2019-01-12:
Or, maybe I do spiOpen twice, one for each channel/mode and use the associated handles from the open?

Instead of closing and reopening.

joan2937, 2019-07-07:
Yes, I would do it as you suggest.  Call SpiOpen  for each needed configuration and use the appropriate handle.

closed 2019-07-07 by joan2937