joan2937/pigpio #259: Cannot start pigpiod: mmap gpio failed

SpectralMemories, 2019-01-22:
Hi,
I know this issue has already been talked about, but I got the following issue:
running sudo pigpiod returns: 
```
initPeripherals: mmap gpio failed (Operation not permitted)
Can't initialise pigpio library
```

I tried adding iomem=relaxed to the boot parameters (which applied correcly as stated by /proc/cmdline), same result. 

Raspberry Pi 3 B+
Arch Linux ARM aarch64
Kernel 4.20.3-1-ARCH
Latest commit (34b33c3bf3c6f302539b9fcd0dfff52bd83a21a5)

Thanks for any feedback

jveber, 2019-01-28:
Same there... using OpenSuse Tumbleweed - AArch64 ofc.

jveber, 2019-02-08:
Found out it is possible to use iomem=relaxed as a workaround for CONFIG_STRICT_DEVMEM=y but it does not resolve all the issues you'll encounter on AArch64 :).
First of all pigpio is not written well for portability 32/64 so some pointers are converted to uint32_t. That of course breaks the program.
Mailbox is not on "/dev/vcio" and also no createable as a char device with ID 100 soo.. it needs workaround using "/dev/vchiq" (at least I think that is the right mbox device.. but I'm not sure...)

The biggest problem is DMA because it is hardvired with 32bit pointers... that is the problem I was not able to resolve (it is a bit more complicated...) so I decided to use another library for 64bit scenario.

mwkroening, 2019-06-07:
@jveber, which library do you mean?
I am as well looking for a way to get GPIO working on arm64.

jveber, 2019-06-11:
Libgpiod https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/ - but it does not support PWM.

pbanasik, 2019-08-15:
Summarizing the current status on my RPi 3B+, aarch64, Ubuntu Bionic 4.15.0-1041-raspi2, pigpio compiled from the repo:
- iomem=relaxed is required to make anything work,
- pigpio works (including i2c and PWM), x_pigpio passes
- pigpiod starts but doesn't work , x_pigpiod_if and x_pigpiod_if2 fail even on the basic tests

I'm happy to help testing any patches for pigpiod

joan2937, 2019-08-15:
Just a note.

I am not providing any 64-bit support as I don't have a 64-bit system and will not have one unless Raspbian defaults to 64-bit.

I might incorporate 64-bit patches prepared by other people but I will not create any.

pbanasik, 2019-08-16:
@joan2937 I know how to fix it but I don't have permission to write to the repo. It's a simple extension of the already merged commit - see my comment here: https://github.com/joan2937/pigpio/commit/55d8b880fca26ccf4b897bca1fe66796b9972345

pbanasik, 2019-08-16:
Patch https://github.com/joan2937/pigpio/commit/d1f3bc81d924d8ff4383f752eedcd094fd3c3949 merged into master. All test cases pass on my system (still using the iomem=relaxed workaround).

akioolin, 2019-08-19:
Hi, All

following the discussion in this thread, I add iomem=relaxed and  strict-devmem=0 in cmdline.txt for my rpi3b+. the kernel version is 4.15.0-1043-raspi2 #46-Ubuntu SMP.
the distribution is ubuntu mate for rpi3b+.  but the pigpiod still report the same error.
the pigpiod version is 64.

any suggestion?

joan2937, 2019-08-19:
@akioolin is this a 64-bit version of Linux?  pigpio version 64 is quite old, do you get the same problem on the latest version?

akioolin, 2019-08-20:
Hi Joan:

Yes, the OS is 64bit for RPi3B+. after remove original pigpiod package and update to V71, the error message is gone. the kernel command line still has  iomem=relaxed and strict-devmem=0. the next step is to connect to real sensor device.

Thank you.

Best Regards,
Akio

guymcswain, 2020-01-03:
Fixed in V71.  See PR #286

closed 2020-01-03 by guymcswain

hugo-a-garcia, 2020-09-23:
Can we reopen this issue please?

As stated in this Ubuntu bug... I can confirm the bug and the fix but this needs to be vetted by someone.

https://bugs.launchpad.net/ubuntu/+source/linux-raspi/+bug/1889926

itatarKeurig, 2020-09-23:
I also see the issue with 32bit Ubuntu 20 server on RP3 and I can confirm the fix works.

guymcswain, 2020-09-23:
> I can confirm the fix (removing PROT_EXEC) works.

You can test this now using the `develop` branch.  To be released soon.

guymcswain, 2020-09-23:
@hugo-a-garcia , this issue is closed and relates to network stack problem on 64-bit.  The PROT_EXEC  is here #375 .